onlaj/Piano-LED-Visualizer #2: visualizer.py error

bo7abib, 2019-05-25:
i have installed all the requirements and follow all the given links and i have tested the given tests and they work properly , but my problem currently is for the visualizer , where i have tried many things to make it work but i get no clues , below is the output log when i tried to sudo the visualizer 

```
> sudo -E python visualizer.py
> no output port
> no light port
> Traceback (most recent call last):
>   File "visualizer.py", line 641, in <module>
>     menu = MenuLCD("menu.xml")
>   File "visualizer.py", line 168, in __init__
>     self.update_songs()
>   File "visualizer.py", line 172, in update_songs
>     songs_list = os.listdir("Songs")
> OSError: [Errno 2] No such file or directory: 'Songs'
> 
```

please guide me what should i do ?

onlaj, 2019-05-25:
Add folder called "Songs" :)

bo7abib, 2019-05-25:
thank you , it's working now but it is only same as lightkeys.py with no difference , what can i do to test the functionality of it before i proceed with the screen part

bo7abib, 2019-05-25:
and why does it show 

no output port
no light port

bo7abib, 2019-05-27:
Thank you very much i have connected the lcd and every thing is working now ,regarding the synthesia i think there is some problem regarding the bluetooth midi becuase it is not appearing any more , i currently waiting for the md-bt01 to connect my pi to the piano so i can make the last test .

onlaj, 2019-05-30:
@bo7abib 
I added "Songs" folder to project.
Yes, RPiMidi is a little moody. Try to restart bluetooth midi service with command "sudo systemctl restart btmidi.service".
I'm assuming you replaced dll files in Synthesia?

bo7abib, 2019-05-30:
@onlaj 
yes i have replaced the .dll and the Bluetooth was showing couple days ago but now they are not visible anymore and i have tried ti install and uninstall synthesia more than 10 times on multiple computers and i was always adding the .dll files and check the "Midi.UseWinRTMidi".

onlaj, 2019-05-30:
@bo7abib 
I think this is the problem with btmidi service. Try to restart your RPi, unpair bluetooth connection and pair it again.

bo7abib, 2019-05-30:
i dont think its from the RPi , because the .dll files will let the synthesia show the bluetooth without RPi
:
[image]

onlaj, 2019-05-30:
There is another workaround which I used before.
You will have to use 3 additional applications. The first one is [loopMIDI ](https://www.tobias-erichsen.de/software/loopmidi.html) which emulates MIDI hardware, the next one is app called [MIDIberry ](https://www.microsoft.com/en-us/p/midiberry/9n39720h2m05?activetab=pivot%3Aoverviewtab) which makes bridge between MIDI bluetooth and hardware port made by loopMIDI and then finally [WindSynthBerry ](https://www.microsoft.com/en-us/p/windsynthberry/9pfq7b9p492x?activetab=pivot:overviewtab) which does the same job as second app but in opposite direction, sending output from Synthesia back to Raspberry Pi.
It is much more struggle to setup everything but the advantage is that I never had any problems with Synthesia not seeing input/output.

bo7abib, 2019-05-30:
i will give a last try as soon as the md-bt01 arrives , then if it still not working i will try this method , one more error coming for me while the scripts run for more than 10-15 minutes or if i choose the screen saver option and then the script canceled:

Traceback (most recent call last):
  File "visualizer.py", line 668, in <module>
    screensaver()
  File "visualizer.py", line 525, in screensaver
    temp = find_between(str(psutil.sensors_temperatures()["cpu-thermal"]), "current=", ",")
KeyError: 'cpu-thermal'

jsm174, 2019-06-03:
@bo7abib, the screen saver uses a TTF font to display the temperature.

Did you install fonts-freefont-ttf?

```
sudo apt-get install fonts-freefont-ttf
```

onlaj, 2019-06-03:
@jsm174 all fonts used in my script are preinstalled in Raspbian OS
@bo7abib  try to add 
`print(str(psutil.sensors_temperatures()["cpu-thermal"]))`
somewhere at the beginning of screensaver function, so I can see what system returns

jsm174, 2019-06-03:
@onlaj 

I ran into this issue because I started with Raspbian Stretch Lite which doesn't have the fonts installed.

Also, I needed to install psutil from pip, ie:

```
sudo pip install psutil 
```

vs

```
sudo apt-get install python-psutil
```

bo7abib, 2019-06-03:
i have tried every thing you guys said but nothing worked , i have solved the problem by removing the temp out of the script , thank you very much

closed 2019-06-07 by bo7abib