onlaj/Piano-LED-Visualizer #3: Program unexpectedly freezes

MrBromaba, 2019-06-04:
Hi, so I have noticed that sometimes while I play the program seems to crash, the LEDs won't light up anymore and the LEDs pressed at the moment of crash stay on. Sometimes this takes a while to kick in but sometimes it happens almost immediately after running the program... any ideas why this happens?

onlaj, 2019-06-04:
What does the console say?

MrBromaba, 2019-06-04:
Actually, nothing. It looks like everything is running well although it doesn't work.

onlaj, 2019-06-04:
Do you have screen connected? Is the menu still working?

MrBromaba, 2019-06-13:
Hi, sorry for the late response. I don't use the display yet as I'm only using the lioghtkeys.py file and I control the Pi with a SSH client. Seems like the problem isn't coming from your code but rather from the Raspi, I sometimes the green LED in goes off and then the LED strip freezes. Hasn't happened in the past couple of days...

closed 2019-07-05 by onlaj