onlaj/Piano-LED-Visualizer #5: Purpose of output port?

jsm174, 2019-06-06:
Looking through the code, it doesn't seem like output port (outport) really does anything at this time. What is its intended purpose?

I have almost everything working, but I am trying to figure out the Synthesia setup.

Synthesia for the iPad is connected to RPi-Bluetooth, and set to visualizer Input. How would I then get the keyboard connected to the Synthesia? 

Do I need to connect the keyboard to Synthesia via a cable?

Can the pi emit two bluetooth midi devices?

onlaj, 2019-06-06:
If your piano have MIDI USB you should connect it to RPi using OTG cable. 
If you want to use Blueooth instead you could connect your piano to RPi and it should work, but I guess your piano is not able to search and pair other devices, right?
You can try to connect your piano to iPad (by bluetooth) and in Synthesia set your Piano as input and both piano and RPi as output. Then you will be still able to hear the sound from your piano when Synthesia is sending midi.
As for output port, this is just leftover from when I tried to make a bridge between devices, before I found out [this solution](https://neuma.studio/rpi-as-midi-host.html) . I kept it because I thought I might need it in future, but I guess I won't use it.

jsm174, 2019-06-06:
My keyboard (Williams Allegro 2) is USB only. 

1) keyboard -> raspberry pi (via usb) with Input set to Williams Allegro 2, all good

2) keyboard -> raspberry pi (via usb) with Playback set to Williams Allegro 2, able to record and play midi files. 

3) iPad -> raspberry pi (via bluetooth) with Playback set to RPi-Bluetooth, able to play midi files to Synthesia

4) iPad -> raspberry pi (via bluetooth) with Input set to RPi-Bluetooth, able to record Midi from Synthesia (however, the midi file sometimes seems garbled)

---

I am going to work more on the Synthesia part tonight. But I think this is how it has to be configured:

1) iPad -> raspberry pi (via bluetooth) with Input set to RPi-Bluetooth, Synthesia Music Output set to RPi-Bluetooth

2) keyboard -> iPad (via camera adapter / usb cable), Synthesia Music Input set to Williams Allegro 2

This way, I will go to play a song in Synthesia, it lights the leds, and waits for input on the keyboard?

I was just hoping I could leverage the RPi, and not have to keep switching the usb cable depending on what I'm actually doing (using synthesia, or just playing the piano) 

---

As far as the RPi bluetooth setup, for the link you shared, I only performed the "Optional Midi Bluetooth Setup" section

```
sudo btmidi-server -v -n "RPi-Bluetooth"
```

I did not do the udev configuration / daemon. Do we really need to do that part?

onlaj, 2019-06-06:
This is how it looks on my setup:
1. keyboard <-> midi to usb adapter <-> Raspberry Pi
2. Windows tablet <-> Raspberry pi (by bluetooth)
When I want to just light keys I set my Piano as input in the Visualizer settings. When I want to Synthesia light my keys I set RPi-Bluetooth as input. I was thinking about making third key as a shortcut to quickly change between those inputs.
In Synthesia I set both Input and Output to RPi-Bluetooth.
You should do everything mentioned on your screenshot beside the readonly part.

jsm174, 2019-06-06:
Okay, I will try that. Thanks! If this setup allows Synthesia to wait for you to press the correct keys on the piano before going to next notes, then awesome!

Two final questions on the instructions link.

1) Did you also do the first part of that article, ie

Configure automatic MIDI connection/disconnection on USB device connect/disconnect

2) If we do the read only part, wouldn't the visualizer script not be able to save midi files to the Song folder?

onlaj, 2019-06-06:
1. yes, it connects your usb midi connections with bluetooth midi.
2. yes, don't do that :D

closed 2019-07-05 by onlaj