onlaj/Piano-LED-Visualizer #6: auto reconnect after piano shutdown

bo7abib, 2019-06-20:
Hello ,

i have been testing the project now for a while and every thing is working very well thanks to the owner of this project , but i am facing currently a small issue which is in this scenario : the RPi and its script is always compiled and turned on but the problem my piano has auto shutdown time when user is inactive maximum to 2 hours so when it turned off i need to cancel the script then turn it on again , so is there any workaround to solve this issue without the need to recompile the script.

onlaj, 2019-06-20:
Do you have screen connected? After restarting your piano Go to menu -> Ports Settings -> Refresh Ports, and then go to Input and choose your piano name.
My piano is connected to RPi through MIDI<->USB adapter, so when I shutdown piano the port is still connected. Even if I reconnect adapter it still works fine, I'm not sure how I can reproduce your problem.

bo7abib, 2019-06-22:
@onlaj 

Thank you very , worked .

closed 2019-06-22 by bo7abib