onlaj/Piano-LED-Visualizer #7: Some LCD screen error for the visualizer.py

Kliklijoker, 2019-07-05:
After I install all the requirements.txt library and connect the LCD screen
I execute the the command sudo -E python visualizer.py, I got following error.

Traceback (most recent call last):
  File "visualizer.py", line 3, in <module>
    import LCD_1in44
  File "/home/pi/Desktop/pianoled/LCD_1in44.py", line 28, in <module>
    import LCD_Config
  File "/home/pi/Desktop/pianoled/LCD_Config.py", line 38, in <module>
    SPI = spidev.SpiDev(0, 0)
IOError: [Errno 2] No such file or directory

Do you know what this error mean and how can I fix it?

onlaj, 2019-07-05:
Try to [enable SPI](https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/)

Kliklijoker, 2019-07-05:
I already set RPi BT as input
Also set synthesia I/O both are RPi BT, but when I using synthesia, the led still doesn lighting and after I change RPi BT as input, the terminal still show my midi device as input.

onlaj, 2019-07-05:
Is BT connection between RPi and Synthesia works at all? When you press keys does Synthesia see it?

Kliklijoker, 2019-07-05:
Update

Now the key will lighting and waiting me press, but when I press the key it still light. And when I press the key, synthesia doesn't have any reaction

onlaj, 2019-07-05:
If you did everything correctly as instructed [here](https://neuma.studio/rpi-as-midi-host.html) there should be two-ways connection between both devices. Make sure you configured ports in Synthesia.

Kliklijoker, 2019-07-05:
What's you mean two ways?
I think the computer set RPi BT as input
But it doesn't receive the signal, are you just plug you piano usb on rpi OTG hub?

onlaj, 2019-07-05:
Yes, you should connect your piano with OTG cable to RPi.
In Synthesia you should set RPI-Bluetooth as input and output.
If you want to light next-to-play keys by Synthesia you should set RPI-Bluetooth as input in Visualizer ports settings. If you want to just light up keys when pressing them you should set your piano as input.
I think this has nothing to do with my code, there must be a problem on Synthesia or [this](https://neuma.studio/rpi-as-midi-host.html) script side.

Kliklijoker, 2019-07-05:
I have a question
Why RPi BT as input can let synthesia know when I press the key?

onlaj, 2019-07-05:
"Why RPi BT as input can let synthesia know when I press the key?"
The [script](https://neuma.studio/rpi-as-midi-host.html) is sending messages from your piano to pc AND from pc to your piano. My script is only for handling lights.

"In the synthesia you say you set RPi BT as I/O , do you have other setting in synthesia?"
No, nothing else.

Try to run "connectall.rb" script manually to see if there are any errors.

Kliklijoker, 2019-07-05:
after I command "connectall.rb"

/usr/local/bin/connectall.rb:12:in `block in <main>': undefined local variable o                                                                                                             r method `names' for main:Object (NameError)
        from /usr/local/bin/connectall.rb:6:in `each'
        from /usr/local/bin/connectall.rb:6:in `<main>'

Issue aconnect -l

client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 20: 'Roland Digital Piano' [type=kernel,card=1]
    0 'Roland Digital Piano MIDI 1'
client 128: 'RPi-Bluetooth' [type=user,pid=1930]
    0 'RPi-Bluetooth   '

[image]

Kliklijoker, 2019-07-05:
If I just play the synthesia the LED will light sometimes white or green, sometimes white or blue(I doesn't know why it not just green and blue).
My RPi-BT (IN) doesn't have any reaction (I already set input as RPi BT) and I already plug into OTG on RPi.

onlaj, 2019-07-05:
Try to replace connectall.rb content with this code:

`t = `aconnect -i -l`
ports = []
names = []
t.lines.each do |l|
  /client (\d*)\:/=~l
  port = $1
  name = $1
 # we skip empty lines and the "Through" port
  unless $1.nil? || $1 == '0' || /Through/=~l
    ports << port
    names << name
  end
end

ports.each do |p1|
  ports.each do |p2|
    unless p1 == p2 # probably not a good idea to connect a port to itself
      system  "aconnect #{p1}:0 #{p2}:0"
    end`
Not sure if this is the case, but it is worth a shot. I basically defined "name" and "names" variables.
btw. can you run "printmidimessages.py" from tests folder and show me example of midi messages from your piano?

Kliklijoker, 2019-07-05:
I already replace the code and command "connectall.rb" I got

/usr/local/bin/connectall.rb:22: syntax error, unexpected end-of-input, expecting keyword_end

onlaj, 2019-07-05:
My bad, I didn't copy last two lines, edited previous comment.

Kliklijoker, 2019-07-05:
note_off channel=0 note=28 velocity=57 time=0
note_on channel=0 note=31 velocity=25 time=0
note_off channel=0 note=31 velocity=82 time=0
note_on channel=0 note=24 velocity=29 time=0
note_off channel=0 note=24 velocity=67 time=0
note_on channel=0 note=28 velocity=36 time=0
note_on channel=0 note=31 velocity=28 time=0
note_off channel=0 note=28 velocity=88 time=0
note_off channel=0 note=31 velocity=65 time=0
note_on channel=0 note=36 velocity=39 time=0
note_off channel=0 note=36 velocity=106 time=0
note_on channel=0 note=30 velocity=41 time=0
note_on channel=0 note=33 velocity=48 time=0
note_on channel=0 note=26 velocity=31 time=0
note_off channel=0 note=26 velocity=101 time=0
note_off channel=0 note=30 velocity=98 time=0
note_off channel=0 note=33 velocity=89 time=0
note_on channel=0 note=31 velocity=43 time=0
note_on channel=0 note=27 velocity=43 time=0
note_on channel=0 note=23 velocity=40 time=0
note_off channel=0 note=27 velocity=87 time=0
note_off channel=0 note=31 velocity=105 time=0
note_off channel=0 note=23 velocity=81 time=0
note_on channel=0 note=29 velocity=46 time=0
note_on channel=0 note=21 velocity=53 time=0
note_on channel=0 note=25 velocity=40 time=0
note_off channel=0 note=21 velocity=80 time=0
note_off channel=0 note=25 velocity=76 time=0
note_off channel=0 note=29 velocity=88 time=0
note_on channel=0 note=21 velocity=41 time=0
note_on channel=0 note=25 velocity=52 time=0
note_on channel=0 note=28 velocity=49 time=0
note_on channel=0 note=29 velocity=66 time=0
note_off channel=0 note=21 velocity=95 time=0
note_off channel=0 note=25 velocity=94 time=0
note_off channel=0 note=28 velocity=100 time=0
note_off channel=0 note=29 velocity=57 time=0

Kliklijoker, 2019-07-05:
OMG, it work!!
but some wired problem, it should right green left blue, but my led sometimes will light right color, sometimes will be white, why?

onlaj, 2019-07-05:
You would have to edit printmidimessages.py (set RPI-Bluetooth as input) and check what messages Synthesia is sending when you play.

Kliklijoker, 2019-07-05:
note_on channel=0 note=36 velocity=47 time=0
note_off channel=0 note=36 velocity=100 time=0
note_on channel=0 note=40 velocity=56 time=0
note_off channel=0 note=40 velocity=96 time=0
note_on channel=0 note=43 velocity=42 time=0
note_off channel=0 note=43 velocity=106 time=0
note_on channel=0 note=34 velocity=52 time=0
note_off channel=0 note=34 velocity=97 time=0
note_on channel=0 note=33 velocity=46 time=0
note_off channel=0 note=33 velocity=108 time=0
note_on channel=0 note=31 velocity=55 time=0
note_off channel=0 note=31 velocity=91 time=0
note_on channel=0 note=38 velocity=67 time=0
note_off channel=0 note=38 velocity=88 time=0
note_on channel=0 note=37 velocity=53 time=0
note_on channel=0 note=40 velocity=56 time=0
note_on channel=0 note=31 velocity=37 time=0
note_off channel=0 note=31 velocity=101 time=0
note_off channel=0 note=37 velocity=127 time=0
note_off channel=0 note=40 velocity=118 time=0
note_on channel=0 note=31 velocity=57 time=0
note_off channel=0 note=31 velocity=102 time=0
note_on channel=0 note=31 velocity=50 time=0
note_off channel=0 note=31 velocity=58 time=0
note_on channel=0 note=36 velocity=53 time=0
note_off channel=0 note=36 velocity=127 time=0
note_on channel=0 note=40 velocity=59 time=0
note_off channel=0 note=40 velocity=93 time=0
note_on channel=0 note=43 velocity=47 time=0
note_off channel=0 note=43 velocity=118 time=0
note_on channel=0 note=48 velocity=61 time=0
note_off channel=0 note=48 velocity=105 time=0
note_on channel=0 note=52 velocity=56 time=0
note_off channel=0 note=52 velocity=121 time=0
note_on channel=0 note=55 velocity=65 time=0
note_off channel=0 note=55 velocity=124 time=0
note_on channel=0 note=60 velocity=85 time=0
note_off channel=0 note=60 velocity=106 time=0

Kliklijoker, 2019-07-05:
it just like I normal plug OTG cable with the note_on and note_off, what should I need to change?

onlaj, 2019-07-05:
I think "light up notes" might be disabled by default in Synthesia, try to change that.

Kliklijoker, 2019-07-05:
Do you mean
[image]

btw, when I set RPi IN as input in synthesia also menu, when I press the key the LED still will light with white color, it shouldn't with it, right

onlaj, 2019-07-05:
I think the order is the same but it would be easier if you would change language to english. First checkbox should be Off, same with last one.

Kliklijoker, 2019-07-05:
[image]

It still sometimes will have white color :(

Another question, what should I set as output when Piano visualizer and Synthesia mode?

onlaj, 2019-07-05:
"For the midi, I select all channel, does it right?"
Yes.
"Another question, what should I set as output when Piano visualizer and Synthesia mode?"
You don't have to change anything in Synthesia, just input in Visualizer port settings.

"My sound have some delay when I play as synthesia learning mode, why?"
This is really weird, there should be no delay because it is your piano playing sound.
Try to restart your RPi after all those changes. Maybe there is something wrong with ports connections.

Kliklijoker, 2019-07-05:
I already restart everything, I find a weird solution.
If I use song I classify myself, it work well, but if I use default song, it will have some bug(LED with white), I thought it cuz the original midi not all in channel as you write.

Kliklijoker, 2019-07-06:
If I want to add new effect in ledsettings, where should I need to change?
I create a new effect in menu, but it always light as white.

onlaj, 2019-07-08:
"And for the solid color select, where I can add or delete color? I can't find in code."
menu.xml

What effect you want to add?

Kliklijoker, 2019-07-08:
The rainbow effect (when I hit the key it will light red to purple from left to right)

onlaj, 2019-07-08:
I added this effect few days ago, update your code :D

Kliklijoker, 2019-07-08:
Does this effect still in the led settings?

onlaj, 2019-07-08:
Yeah, you need to update visualizer.py and menu.xml

Kliklijoker, 2019-07-08:
Okay, so what is this effect name you called, do I just need to following the form like fading or velocity effect?

onlaj, 2019-07-08:
It is under "Led Strip Settings" -> "LED Color" -> "Rainbow Colors".
Under that menu you can configure how your rainbow looks. You have to press "enter" button on one of the options to activate rainbow effect.
You can combine it with "Fading" or "Velocity" mode.

Kliklijoker, 2019-07-08:
Okay, so this effect just as you use for the animation, right?

onlaj, 2019-07-08:
No, keys lights up with rainbow effect when you play.

Kliklijoker, 2019-07-08:
Can you clearly point out which line in your code you create the rainbow effect?

onlaj, 2019-07-08:
I just noticed that I forgot to commit changes in "menu.xml", updated now.
Check last two commits in which I added this feature.

Kliklijoker, 2019-07-08:
Thank you :D , so the output port seems doesn't have any function here, I want to delete all of them in menu.xml and visualizer.py, where can I delete in visualizer.py?

onlaj, 2019-07-08:
What do you want to delete? Output port? Just delete it from menu.xml so you will not see it in menu.

Kliklijoker, 2019-07-08:
Okay, thank you, here is my small question I only left.
As synthesia mode (input set as RPi BT)  it can work well in the midi song in the same channel and separate manual by left and right by me, if the midi song are use separate channel in left and right, the LED will not light in correct color (green and blue), do you have the same problem?

onlaj, 2019-07-08:
No, it works fine for me, Synthesia sends midi messages with proper channel numbers.

Kliklijoker, 2019-07-08:
I think it is because the midi channel, for you code I found you set channel 11 as right, channel 12 for left, do you also set the same channel in synthesia? Or it can be different channel?

onlaj, 2019-07-08:
https://www.synthesiagame.com/forum/viewtopic.php?p=43585#p43585

Kliklijoker, 2019-07-08:
Thanks! That's what I need, after I check the built in songs in synthesia, it arrange channel 3 for right, channel 4 for left, I thought I need to change this to 11&12 for the code can work properly. Or can I just change channel 11&12 to 3&4 for make code work?

onlaj, 2019-07-08:
Second option would be better, but anyway it is weird, Synthesia should set channels 11 and 12 for left and right hands.

Kliklijoker, 2019-07-08:
OK, thank you again, sorry for me make some mistake.
[image]

In built-in songs, channel 1 for the right and channel 2 for the left, do you know how can I change synthesia setting to make channels 11 and 12 for left and right hands?

Kliklijoker, 2019-07-15:
Sorry for another question, you say need to check the printmessages.py when I set RPi-BT as input, so what channel should show in there? 

note_on **channel=0** note=36 velocity=47 time=0
note_off **channel=0** note=36 velocity=100 time=0

When I use synthesia, does it should automatic set right as channel 11, left as channel 12?
(btw, after I chnage the code channel, it still have some bugs)

lazycreek, 2019-07-21:
> Try to [enable SPI](https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/)

Hello Onlaj! First of all thanks for your guide it's total cool!
 I'm currently trying to replicate your setup. 
But it's very hard for me because I'm totally new to RPi and Python programming.

I have the same issue as Kliklijoker had by running the visualizer.py:
Traceback (most recent call last):
  File "visualizer.py", line 5, in <module>
    import LCD_1in44
  File "/usr/led_piano/Piano-LED-Visualizer/LCD_1in44.py", line 30, in <module>
    import LCD_Config
  File "/usr/led_piano/Piano-LED-Visualizer/LCD_Config.py", line 40, in <module>
    SPI = spidev.SpiDev(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

SPI is already enabled - so I don't have any idea whats wrong. Do I have to install anything else? 
I don't understand where "LCD_lin44" for example will be imported...

closed 2019-07-22 by Kliklijoker

omidfk, 2019-11-10:
> Try to [enable SPI](https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/)

SPI is enable,but I execute the the command sudo -E python visualizer.py, I got following error.

Traceback (most recent call last):
File "visualizer.py", line 3, in
import LCD_1in44
File "/home/pi/Desktop/pianoled/LCD_1in44.py", line 28, in
import LCD_Config
File "/home/pi/Desktop/pianoled/LCD_Config.py", line 38, in
SPI = spidev.SpiDev(0, 0)
IOError: [Errno 2] No such file or directory