onlaj/Piano-LED-Visualizer #8: Delay between key press and LED light on

binokaryg, 2019-07-12:
Previously, I had tried this using Raspberry Pi B (older) and everything was working fine (no noticeable delay).
When I recently tried it with a new Raspberry Pi 3B+, I noticed that there is often a delay between key press and LED lighting up. The delay is more noticeable when I press a sequence of keys in a short time. I am using `lightkeys.py` to test it.
`printmidimessages.py` seems to do fine, although it is difficult to notice any delay with it.
What may be the reason for the delay in the LED? Any idea?

onlaj, 2019-07-13:
Try to run [cProfile](https://docs.python.org/3/library/profile.html#module-cProfile) to check which part of code makes the delay.

binokaryg, 2019-07-13:
I think there is a delay when registering that a key is pressed. The LED works fine when I load it from a MIDI file but a delay is seen during fast notes when the same MIDI is used with Synthesia -> Bluetooth -> LED. Any suggestions on how to reduce this delay? Closing this issue because it might not be due to the problem with your code.

closed 2019-07-13 by binokaryg