onlaj/Piano-LED-Visualizer #9: MidiPorts instance has no attribute 'inport'

AbrahamWisman, 2019-07-25:
Hi, Im quite new to RPI and MIDI and trying to run the visualizer. But get a error;

1. `$ sudo -E python visualizer.py`
2. Traceback (most recent call last):
3.   File "visualizer.py", line 733, in <module>
4.     midipending = midiports.inport.iter_pending()
5. AttributeError: MidiPorts instance has no attribute 'inport'

I have synthesia on my laptop connected with bluetooth, i can send midi and i can see with the bluetooth test (`sudo btmidi-server -v -n "RPi Bluetooth" `) that i receive data on my rpi.

I dont really see a solution for this problem atm. can you help me?

onlaj, 2019-07-25:
Hello.
Update your repository to the latest version, just in case.
Type `aconnect -l` in console and paste the response here.
Try to run visualizer.py and this time copy and paste whole console output.

AbrahamWisman, 2019-07-25:
`aconnect -l ` give's me the following response when i run `sudo btmidi-server -v -n "RPi Bluetooth"` in another terminal.

```
1. pi@raspberrypi:~ $ aconnect -l
2. client 0: 'System' [type=kernel]
3.     0 'Timer           '
4.     1 'Announce        '
5. client 14: 'Midi Through' [type=kernel]
6.     0 'Midi Through Port-0'
7. client 128: 'RPi Bluetooth' [type=user,pid=1943]
8.     0 'RPi Bluetooth   '
```

and this response when i am just connected with my laptop

```
1. pi@raspberrypi:~ $ aconnect -l
2. client 0: 'System' [type=kernel]
3.     0 'Timer           '
4.     1 'Announce        '
5. client 14: 'Midi Through' [type=kernel]
6.     0 'Midi Through Port-0'
```

This is the whole output when trying to run visualizer

```
1. pi@raspberrypi:~/Downloads/Piano-LED-Visualizer-master/Piano-LED-Visualizer-master $ sudo -E python visualizer.py
2. Traceback (most recent call last):
3.   File "visualizer.py", line 733, in <module>
4.     midipending = midiports.inport.iter_pending()
5. AttributeError: MidiPorts instance has no attribute 'inport'
```

onlaj, 2019-07-25:
So you don't have connected keyboard to RPI, right? I will make an exception so it won't crash when there is no input set.

AbrahamWisman, 2019-07-25:
I dont have a piano midi keyboard connected to my rpi.

I do have a computer keyboard connected

onlaj, 2019-07-25:
By default it tries to set keyboard as input, if you want to set your PC as input you have to run visualizer.py and then change input in menu. I just made an update, download new version.

AbrahamWisman, 2019-07-25:
Thanks!
When i run the updated version, there is no output/response. Nothing happends

onlaj, 2019-07-25:
Yes, because there are no ports set yet.
Go to "Ports Settings" -> "Refresh ports"
and then to "Input" and choose RPI-Bluetooth.
Now the keys should light up when you play midi files with Synthesia (if you configured everything correctly in Synthesia)

AbrahamWisman, 2019-07-25:
I mean when i try to run the updated version.

```
1. pi@raspberrypi:~/Downloads/Piano-LED-Visualizer-master $ sudo -E python visualizer.py
2. 
3. 
4. 
```

This is the response when i press keyboard interrupt after some minutes
```

1. ^CTraceback (most recent call last):
2.   File "visualizer.py", line 733, in <module>
3.     menu.show()
4.   File "visualizer.py", line 341, in show
5.     self.LCD.LCD_ShowImage(self.image,0,0)          
6.   File "/home/pi/Downloads/Piano-LED-Visualizer-master/LCD_1in44.py", line 310, in LCD_ShowImage
7.     pix = pix.flatten().tolist()
8. KeyboardInterrupt
9. 

```

onlaj, 2019-07-25:
What exactly you are trying to achieve? You don't have a keyboard connected to RPI so I guess you want to light up LEDs with Synthesia, right?

AbrahamWisman, 2019-07-25:
Yes, i want to play a midi file on my laptop on synthesia, send the midi data over bluetooth to rpi and light up LEDS.

onlaj, 2019-07-25:
Do you have screen on your RPI? If yes then refresh the ports and then set RPI-Bluetooth as input. Then it should work. This is how it looks when I disconnected the keyboard to recreate your setup: https://gfycat.com/pl/shockedcarefreebighornedsheep

AbrahamWisman, 2019-07-25:
Ah, i dont have the small LCD screen, instead i have a 27 inch HDMI screen connected to it :) can i use that?

onlaj, 2019-07-25:
No, I'm afraid you will have to edit the visualizer.py :D 
Replace line 665 in latest version with
`if "RPi" in port:`
It will set RPI-Bluetooth device as input.

AbrahamWisman, 2019-08-02:
Thanks a lot, that works!
I also have a midi cable connection. I can see that the bluetooth midi data is not as steady as the midi cable. All the data is correct, but there is a delay of +-half a second sometimes.

closed 2019-08-23 by onlaj