onlaj/Piano-LED-Visualizer #10: LCD Display is getting white

lazycreek, 2019-07-26:
Hi Onlaj,

first thanks for your guide!

Finally I managed to connect the keyboard (Yamaha P-45B) with the Raspberry Pi and Synthesia.
Synthesia is waiting for me to press the keys and the LEDs light up accordingly.

However, I still have 2 small problems/questions, maybe you can help me.

First my setup:
Raspberry Pi 3 Model B Rev 1.2
LCD Display Waveshare Raspberrypi 128x128 (the same one you use)
WS2812B LED Strip currently only 1 meter (I'll change it soon)

Problem one:

As soon as I call sudo -E python visualizier.py as pi user, my screen becomes completely white. Before that it shows the normal Linux console.

Can you explain to me how I get the menu shown on the display where you select the songs? 
How do I get this displayed after the Raspberry Pi restart?

I don't know if it's because I was forced into Pythonscript: LCD_Config.py make an adjustment: 
from SPI = spidev.SpiDev(0, 0) to SPI = spidev.SpiDev(0, 1)

The reason for this is that I don't have /dev/spidev0.0... i just don't know why

Problem two:

I would like the LEDs to light up before I press the buttons. 
At the moment they light up together with the keystroke.

pi@raspberrypi:/usr/led_piano/Piano-LED-Visualizer $ aconnect -l
client 0: 'System' [type=Kernel]
    0 'Timer '
    1 'Announce'
client 14: 'Midi Through' [type=Kernel]
    0 'Midi Through Port-0'
client 20: 'Digital Piano' [type=Kernel,card=1]
    0 'Digital piano MIDI 1
        connected to: 128:0
        connected from: 128:0
client 128: 'RPi-Bluetooth' [type=User,pid=569]
    0 'RPi-Bluetooth '
        connected to: 20:0
        connected from: 20:0

If i run the printmidimessages.py it shows a lot of "clock time=0".. 
pi@raspberrypi:/usr/led_piano/Piano-LED-Visualizer/tests $ sudo -E python printm idimessages.py
Inport set to Digital Piano:Digital Piano MIDI 1 20:0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
note_on channel=0 note=59 velocity=49 time=0
clock time=0
note_on channel=0 note=60 velocity=49 time=0
clock time=0
clock time=0
note_on channel=0 note=53 velocity=38 time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0
clock time=0

Thanks in advance!

onlaj, 2019-07-27:
> As soon as I call sudo -E python visualizier.py as pi user, my screen becomes completely white. Before that it shows the normal Linux console.

It should be white from the beginning. Which version of Linux you have installed?

> The reason for this is that I don't have /dev/spidev0.0... i just don't know why

Same as above, I'm guessing it is because OS.

> I would like the LEDs to light up before I press the buttons.

You mean lighting next-to-play keys with Synthesia? You have to change input using Visualizer menu, but since your screen doesn't work you can't do it.

Try to install Raspbian.

lazycreek, 2019-07-28:
Hi,
I installed the newest version of Raspbian Buster full date 2019-06-20 

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I will try a new installation and check after SPI enabling if /dev/spidev0.0 is available.

lazycreek, 2019-07-28:
Hi,

I now installed the Raspbian Buster Lite and installed every library / bluetooth and so on. 
Now the display is working just like you suspected. yay :)

Can you tell me where I have to do the changes for "next to play keys"? in vizualizer menu? 
Thanks again for your help!

onlaj, 2019-07-28:
Ports Settings -> Input -> RPI-Bluetooth
If there is no BT device try to refresh ports on previous page.

closed 2019-08-23 by onlaj