onlaj/Piano-LED-Visualizer #13: Usb midi pass through?

Tallis161, 2019-08-21:
Hi, thanks for all your work :)
I'm pretty sure I'll have it working once I select the input port, I haven't soldered the pins needed for the screen yet, I think setting it to Bluetooth by default will work as it did for another open issue.

As latency is a concern I want to try using direct usb midi, but I also use the midi for a vst instrument on my laptop (garritan cfx :D) 
So I'm wondering can I set my Yamaha piano as midi input then output from the rpi (over a male to male usb cable) to my laptop?

Tallis161, 2019-08-21:
Just to clarify I got the same error about the midi input port, connected via pc trying to test using a software midi keyboard that I could see was sending signal via btmidiservice -v 

So I'm pretty sure it'll work, I just need to actually connect my piano, though hopefully with the screen or that override in line 650ish to set Bluetooth input it'll work fine.

I was thinking it might help to add to instructions to make sure libjpeg and numpy are installed, though it's possible I missed that somewhere.

Tallis161, 2019-08-22:
OK I have the screen working, it displays the menu once visualiser is started, though none of the buttons are working.
I'm assuming that'll be because the display pins are soldered properly but not the control pins, though if there's another possible cause thought I'd mention..
I feel very close to having it function normally.. Again you're a legend for putting this all up on github. I wouldn't have had a clue without the guide.

Tallis161, 2019-08-25:
Deleted my spammy comments, so Id still like to work out the usb midi pass through, but I think I figured out how to get midi into the pi from Windows (the other way around from what I want to do) by using two generic midi usb interfaces, pass through on Windows from piano and hopefully not much latency.

I am Actually having the freezing issue though, I have my pi running with the TV as a screen but no other code turns up, the leds just stop responding. After interrupting it I just got a trace back to lines 999, in module led. Strip. Strip.
Neopixel.py line 110 in show respect = ws. Ws2811_render (self. Leds)

I wanted to suggest adding to the instructions to do apt-get libjpegdev, python jumpy and spidev,  though they seem obvious instructions are best if fool proof I think :) 
And finally would it be possible to make a colour mode using velocity to determine colour on a spectrum, like blue to red? Like cubase lol. 

The only other idea I had is a light mode that also turns on the the led either side at maybe 25% brightness, like a lens flare kind of thing? Might be terrible.

Also any way to set this up for a synth with fewer octaves?

onlaj, 2019-08-26:
Hey.
Under my youtube video user Frederic Vendetti suggested using midi splitter if you want to connect more midi devices, I think it is a brilliant idea, maybe give it a try?

I'm afraid I don't have enough information to help you with the freez problem :/

"apt-get libjpegdev, python jumpy and spidev"
I think it is already installed if you are using Raspbian, I might add a suggestion which OS to choose.

I'm currently working on some effects, thanks for suggestions.

"Also any way to set this up for a synth with fewer octaves?"
I guess you just have to change led count, it should work.

Tallis161, 2019-08-26:
Ok the freeze seems more to be a midi communication issue, like another user I'm getting sticky keys, I think it's the Bluetooth midi honestly. Which could hopefully be solved by the midi splitter solution. 

I do think it's a great idea, though because my piano is usb out I can't find a splitter, maybe it could work by using a usb midi to the piano then a splitter from there, and two more usb midi interfaces. 
It's an awful last resort though, so I'm hoping a more direct solution works.. Apparently midi is a one way even with usb, so I'm literally just soldering another usb A plug onto a cable. Fingers crossed..

Looking forward to your new effects :)
I'll try just changing the led count, I thought the range being different might have an effect, as it wouldn't be starting from A1 with fewer leds. I'll try it out, maybe it is that easy.

Tallis161, 2019-08-27:
Ok apparently although midi is one way it's still a loop, and the midi usb interface uses an A plug not a B, so kind of back to square one.
 I was kind of hoping with rtmidi I can input to the pi first then pass through via male to male usb A, it would solve the issue of note off messages not getting over Bluetooth properly. I'm going to try it when I can get a cable.

Tallis161, 2019-08-28:
I've gone back to just using usb midi into the pi, and it's freezing every few minutes. What kind of info would help diagnose that? I'm going to move my display back so I can see the code without SSH but the last time I did that it gave the regular exit code. It works great when it works but I'm not sure what I changed, it was fine yesterday and I've been using read only 0.o

Tallis161, 2019-09-07:
I now can't get it to run for more than maybe 10-15 minutes before it freezes, on either usb in or bluetooth in.

Tallis161, 2019-09-19:
And I had another idea for getting a pass through working. It seems to struggle sending midi via Bluetooth,

Could I add lines to transmit the serial midi data to a GPIO pin?

I can then wire it to a midi usb converter that will pass the signal on as usb to another device.

onlaj, 2019-09-20:
> Any solve you know of for this?

Try to use [screen](https://linuxize.com/post/how-to-use-linux-screen/)

> Could I add lines to transmit the serial midi data to a GPIO pin?
Well, I guess it could work but I'm currently working on different solution, still waiting for the parts to be delivered.

Tallis161, 2019-09-20:
Thanks, screen looks great.

Different solution sounds interesting, if you don't mind me asking what sort of parts/how would it work?

onlaj, 2019-09-20:
RPI <-> USB HUB <-> USB TO MIDI Interface <-> midi female to female adapter <-> MIDI to USB Interface <-> Tablet/PC
It should work like USB to USB connection with midi interface in between.

Tallis161, 2019-09-20:
Nice, the usb midi cables are useful for weird solutions like this

Tallis161, 2019-09-20:
I think I might try it actually, I can use my two midi usb pcbs and just connect the two serial parts by wires. Makes it a little smaller too :)

Tallis161, 2019-09-21:
soo it seems to be the pi struggles to output to a midi usb converter, or at least the one I have. 
But, I tried the midi pass through on Mac I was using, this time using the usb midi, and it works. seems solid, shouldnt have the sticky keys issue.

I'm not sure how I didn't think of this, I went into a whole thing about arduino usb host shields, buffer circuits and then the usb midi pcb, but It didn't occur to me to just hook two up together. 
Brilliant. The positives aren't mirrored, so just connecting in +/- to out +/- and the common ground, and it works :)
[image]

Tallis161, 2019-09-21:
boy did I assume wrong. seems its not bluetooth's fault at all.
No idea why but still getting sticky keys. maybe the pi zero can't keep up with too many note off messages coming in at once?

Tallis161, 2019-09-22:
Ok, what I think I learned, you are actually better off using your female to female midi adapter, sure you can make it a single smaller cable with the two enclosures taped together, but some of the connection pads on these cheap things are just dodgy, I tried swapping around the input/.output of the board, so the board that was inputting into the pi would be outputting to the pc, but It didn't work, I think maybe just because of bad soldering, like one output + port just had no copper to add solder to, nothing would stick. so cheap pcbs are generally made cheaply. one even neglected the midi serial ground entirely. 

So I went back to the cable I used last night, output from PC into the pi, and I found the sticky keys issue I think is to do with Mac OS, or the particular app, midi pipe.
I'm using midi re-link on windows, and it seems to be better at least, I havent noticed any sticky keys but i honestly think it's just trying to lull me into a false sense of security so it can get me worse later.
Still, some progress, some new things learned.

closed 2020-02-22 by onlaj