onlaj/Piano-LED-Visualizer #14: I tried to auto run your script every time the Raspberry boots but I get this error message...

MrBromaba, 2019-09-08:
I followed these instructions to auto run your script: https://www.raspberrypi-spy.co.uk/2015/02/how-to-autorun-a-python-script-on-raspberry-pi-boot/

Used the 'A Script Without End' part because your script uses an endless loop. When I reboot, I get this error message:

 Traceback (most recent call last):
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 641, in <module>
    menu = MenuLCD("menu.xml")
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 161, in __init__
    self.DOMTree = minidom.parse(xml_file_name)
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 1918, in parse
    return expatbuilder.parse(file)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 922, in parse
    fp = open(file, 'rb')
IOError: [Errno 2] No such file or directory: 'menu.xml'

However, the script runs perfectly fine if I manually start it. Do you know how I could make it work?

MrBromaba, 2019-09-20:
Well, doesn't your code include all the examples from the Neopixel library which brighten up all the LEDs? So far, my stripe / Raspberry hasn't burnt yet :D however, if you don't want to include it in your code, could you maybe explain how I could include it in my own version to make the brightness dimmable with the screen? Everything that's more advanced than defining a new variable is above my level of coding, lol...

onlaj, 2019-09-22:
[image]
I will commit tomorrow after some testings.

MrBromaba, 2019-09-22:
Cool, looking forward to it. Thanks for all the work man!!

MrBromaba, 2019-09-28:
Hey, I saw you committed, but only the brightness function. Any chance you could add a backlight mode as well (as I described above)?

onlaj, 2019-10-07:
Sorry for late response, I didn't get any notification from Github.
Yeah, I totally misunderstood you. I will add this option in next update, maybe tomorrow.

onlaj, 2019-10-07:
[image]
Is it something you wanted? It doesn't work with "Fading" or "Velocity" mode tho. I will commit tomorrow.

MrBromaba, 2019-10-07:
Heya, yeah something like that. How can I commit my changes to the repository? Sorry, I'm a noob at this topic 😅. I basically only added a for loop before line 982 that runs through each led and illuminates it with the backlight value. Then later in the code, where you define to turn off the leds again after the key has released, I changed it to the backlight value. It also works in fading and velocity mode for me.

onlaj, 2019-10-07:
Yeah, I thought it will be that simple and did something similar but when I enter "fading" mode it starts to blink very fast. I will make a closer look tomorrow. Can you upload the code to pastebin so I can see what exactly you changed?

closed 2020-01-06 by onlaj