onlaj/Piano-LED-Visualizer #15: Do anybody have ready iso file with this project?

bitterspy, 2019-09-25:
Hi, im struggling with lots of errors and have no time to do with this project anymore. Do anybody have working system and is able to share just ready to download iso image from SD card? If yes please share with me. I will be very gratefull. Thanks

vzoltan, 2020-02-18:
Hi, im also interested in a working img file.

Could someone pls dd his/her sd card and share?

konradqwerty, 2020-02-18:
Try PianoTutor, I also couldn't run Onlaj's

vzoltan, 2020-02-18:
thx, bud id like to use my pi zero + led setup.

vzoltan, 2020-02-20:
Ill will share mine.
almost done, everything working, except i cant see my rpi-bt in synthesia on my windowss.
the leds are on while i play on the piano, now i need to find a fix for my synthesia connection or bt connection problem, and then ill share the image.

rickstaa, 2020-06-11:
I created a backup of my working img using dd it can be found [here](https://sourceforge.net/projects/piano-led-visualizer/). Let me know if it works.

closed 2020-12-23 by onlaj