onlaj/Piano-LED-Visualizer #17: connecting wires to GPIO pins

glootkoopshire, 2019-10-07:
Hi, I have gotten the code to work, I am just having trouble getting the cables connected to the GPIO pins. Do you connect the cables before or after you install the LCD hat? When you say wrap, do you mean just curling it around the pin with your finger, or using a wire wrap tool? do you know any tutorials I should watch in order to do this? i am very lost

onlaj, 2019-10-07:
Yes, I just wrapped them with my fingers. I didn't find any 'proper' solution so I improvised.
This is how it looks:
[image]

Just make sure wires doesn't touch other pins.

glootkoopshire, 2019-10-07:
so when you wrap it around, do you just take the strands and weave them together or just leave them touching like that? how tight against the pin does it have to be? sorry i am noob with stuff like this xD

onlaj, 2019-10-07:
I put heat shrink bands over and heat them a little. As long as it doesn't touch other pins it should be fine.

ajayprk, 2019-10-09:
I used a GPIO 1 to 2 expansion board to keep it easy

closed 2019-10-10 by glootkoopshire