onlaj/Piano-LED-Visualizer #18: Running Visualizer.py

harryG15, 2019-10-07:
Hi,
Just having a problem with running the visualizer. I went into library ' cd Python-LED-Visualizer ' then ran ' sudo -E python visualizer.py '  but when running it said 'missing module, import rtmidi' - something along those lines? Have i cloned correctly? And so i need to be in ' Read only Mode for this to work?' ive tried this numerous times becuase my sd sometimes crashes after rebooting. Any thoughts would be appreciated..
Thanks
Harry

onlaj, 2019-10-07:
Did you install all required libraries? You can find them in "requirements.txt" file.

> And so i need to be in ' Read only Mode for this to work?'

No, you skip that part.

harryG15, 2019-10-07:
Great - thanks for the reply! Yes I do believe I installed all the required ones - but i will double check! i used ' sudo pip install RPi.GPIO ' for example.. through all of them? 
The only other thing i think it could be - is something to do with the 'automatic midi connection part' as i reckon the SD card corrupts for some reason when i set up midi connection on boot?
I'll get back to you with an update - how i get on - it might just be me?!
Thanks

onlaj, 2019-10-07:
You can install them with one command using:
`pip install -r requirements.txt`

harryG15, 2019-10-07:
Ok, great. I'll let you know how i get on
Thanks
Harry

harryG15, 2019-10-08:
Hi again, I’m just wondering if I need to do the ‘Set Rpi and Bluetooth midi host’ part of your instruction - as I don’t want to use Synthesia or a connect to a computer - but want to JUST plug Rpi and keyboard in with a USB connecting them - and have Visualizer running?
Is that possible? 
Thanks
Harry

alxork, 2019-10-08:
Hello, same here, i'd like to just be able to have the leds light when I play the piano, no need to have the synthesia running. Can we just skip the Midi-bluetooth setup part?
Thank you!
Alex

onlaj, 2019-10-08:
@harryG15 @alxork you don't need a PC to light up keys

harryG15, 2019-10-08:
that's brilliant - thank you - got it working!!! :) 
One other point is - turning it on or off... How do i automatically set the visualizer.py on when i plug in the power... or do i have to manually start it up with keyboard and mouse? 
Or do i leave the power plug in the whole time? (Probably not?)

onlaj, 2019-10-08:
> Or do i leave the power plug in the whole time? (Probably not?)

Just leave it running, it doesn't consume much power. After few minutes my script goes to low performance mode by changing screen update rate. 
As for running script on boot, check this https://www.raspberrypi-spy.co.uk/2015/02/how-to-autorun-a-python-script-on-raspberry-pi-boot/

harryG15, 2019-10-08:
I keep getting glitches whenever I do it and the lights are also random?! :(

onlaj, 2019-10-08:
You need endless loop.
Make sure you have correctly connected the cables.

ajayprk, 2019-10-09:
Do you recommend setting the config to read only after everything works to avoid corruption? I read it on the link you posted for setting up USB midi host.

onlaj, 2019-10-09:
> Do you recommend setting the config to read only after everything works to avoid corruption? I read it on the link you posted for setting up USB midi host.

You can do it but you won't be able to record midi.

harryG15, 2019-11-01:
Hi again, after a while I managed to get it all working - so all good!
Question though, I have the lights connected with a 'Midi IN/OUT to USB' cable into RPi... ( which obviously connects the two...) 
I also want to record the midi (being played) through logic pro X (live when playing) for audio! 
should I get a:
1. USB splitter so the lights can work but also record live into logic pro simultaneously on MacBook (don't think this works??)
2. send the midi recorded from THE RPi!! onto my MacBook using a transfer? 
I need the midi file for a youtube video ( setting a piano sound to it etc...) (if you follow...)

Any help, or direction would be appreciated!
Harry

bradliegek3, 2020-01-01:
> Hi again, after a while I managed to get it all working - so all good!
> Question though, I have the lights connected with a 'Midi IN/OUT to USB' cable into RPi... ( which obviously connects the [...]

Hi Harry, can you tell me how to managed to get it all working with the automatic boot of visualizer?

closed 2020-02-22 by onlaj