onlaj/Piano-LED-Visualizer #19: Python-rtmidi error

Shahroozgr, 2019-10-10:
Hello
I do the installation process in the following order:
First of all, I do the LED connection and install the libraries and activate SPI as your instruction. And after that, configuring automatic midi connection as your instruction.
Of course, I skip  BLUETOOTH MIDI settings for two reasons (I don't need it and I get a lot of errors in launching it).
After the above steps, I clone the Visualizer in my Raspberry Pi 3B+ and Using pip install all libraries listed in requirements.txt file but python-rtmidi is not installed and I get the following error message:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python-rtmidi is not available, but is referred to by another package 

I tried other ways but I got another error message:
E: Package 'python-rtmidi' has no installation candidate

Could you please guide me where I am wrong and how to fix it.
regards

onlaj, 2019-10-10:
I think it happens because you skipped the bluetooth midi connection, try install those packages

```
sudo apt-get install -y autotools-dev libtool autoconf
sudo apt-get install -y libasound2-dev
sudo apt-get install -y libusb-dev libdbus-1-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev
```

Shahroozgr, 2019-10-10:
thanks for your answer
I tried it before but there is an error in this command:
sudo apt-get install -y libusb-dev libdbus-1-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev
that can't be install.

onlaj, 2019-10-10:
What errors you exactly see? Can you paste them here?

Shahroozgr, 2019-10-10:
I will to try it again and paste the error here in half an hour
thanks so much

Shahroozgr, 2019-10-10:
I got these error during installation:

1-When I want to test the auto-connectin with command connectall.rb I get this massage:
Traceback (most recent call last):
	2: from /usr/local/bin/connectall.rb:6:in `<main>'
	1: from /usr/local/bin/connectall.rb:6:in `each'
/usr/local/bin/connectall.rb:12:in `block in <main>': undefined local variable or method `names' for main:Object (NameError)

But when I check the result with aconnect -l, I can see my MIDI keyboard:
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
client 14: 'Midi Through' [type=kernel]
    0 'Midi Through Port-0'
client 20: 'SL STUDIO' [type=kernel,card=1]
    0 'SL STUDIO MIDI 1'
    1 'SL STUDIO MIDI 2'

2-In MIDI Bluetooth setup, after clone the oxesoft/bluez in “make” command I got this error:
pi@raspberrypi:~/bluez $ Make
bash: Make: command not found
 But after that error “sudo make install” is installed.

3-In pip install -r , all my package are installed but for python-rtmidi I got this error:
pi@raspberrypi:~ $ pip install -r /home/pi/Piano-LED-Visualizer/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (from -r /home/pi/Piano-LED-Visualizer/requirements.txt (line 1)) (0.6.5)
Requirement already satisfied: webcolors in ./.local/lib/python2.7/site-packages (from -r /home/pi/Piano-LED-Visualizer/requirements.txt (line 2)) (1.10)
Requirement already satisfied: psutil in ./.local/lib/python2.7/site-packages (from -r /home/pi/Piano-LED-Visualizer/requirements.txt (line 3)) (5.6.3)
Requirement already satisfied: mido in ./.local/lib/python2.7/site-packages (from -r /home/pi/Piano-LED-Visualizer/requirements.txt (line 4)) (1.2.9)
Requirement already satisfied: Pillow in /usr/lib/python2.7/dist-packages (from -r /home/pi/Piano-LED-Visualizer/requirements.txt (line 5)) (5.4.1)
Collecting python-rtmidi (from -r /home/pi/Piano-LED-Visualizer/requirements.txt (line 6))
  Using cached https://files.pythonhosted.org/packages/af/fb/705cbdc6f27bdb428f57f4751034665798ca166fad7acb57a28a148a1549/python-rtmidi-1.3.0.tar.gz
Requirement already satisfied: rpi-ws281x in /usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg (from -r /home/pi/Piano-LED-Visualizer/requirements.txt (line 7)) (1.0.0)
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from webcolors->-r /home/pi/Piano-LED-Visualizer/requirements.txt (line 2)) (1.12.0)
Building wheels for collected packages: python-rtmidi
  Running setup.py bdist_wheel for python-rtmidi ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-QCVkmm/python-rtmidi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-Sj5S0M --python-tag cp27:
  Package jack was not found in the pkg-config search path.
  Perhaps you should add the directory containing `jack.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'jack' found
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  creating build/lib.linux-armv7l-2.7/rtmidi
  copying rtmidi/release.py -> build/lib.linux-armv7l-2.7/rtmidi
  copying rtmidi/midiutil.py -> build/lib.linux-armv7l-2.7/rtmidi
  copying rtmidi/__init__.py -> build/lib.linux-armv7l-2.7/rtmidi
  copying rtmidi/midiconstants.py -> build/lib.linux-armv7l-2.7/rtmidi
  running build_ext
  building 'rtmidi._rtmidi' extension
  creating build/temp.linux-armv7l-2.7
  creating build/temp.linux-armv7l-2.7/src
  creating build/temp.linux-armv7l-2.7/src/rtmidi
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-9NJ3qw/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -D__LINUX_ALSA__ -D__UNIX_JACK__ -Isrc/rtmidi -I/usr/include/python2.7 -c src/_rtmidi.cpp -o build/temp.linux-armv7l-2.7/src/_rtmidi.o
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-9NJ3qw/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -D__LINUX_ALSA__ -D__UNIX_JACK__ -Isrc/rtmidi -I/usr/include/python2.7 -c src/rtmidi/RtMidi.cpp -o build/temp.linux-armv7l-2.7/src/rtmidi/RtMidi.o
  cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
  src/rtmidi/RtMidi.cpp:2935:10: fatal error: jack/jack.h: No such file or directory
   #include <jack/jack.h>
            ^~~~~~~~~~~~~
  compilation terminated.
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for python-rtmidi
  Running setup.py clean for python-rtmidi
[... 10 lines cut ...]
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/rtmidi
    copying rtmidi/release.py -> build/lib.linux-armv7l-2.7/rtmidi
    copying rtmidi/midiutil.py -> build/lib.linux-armv7l-2.7/rtmidi
    copying rtmidi/__init__.py -> build/lib.linux-armv7l-2.7/rtmidi
[... cut ...]
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-QCVkmm/python-rtmidi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-pKoYCI/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-QCVkmm/python-rtmidi/

ajayprk, 2019-10-10:
I had the same exact error and did a lot of googling to sort it out. Here is what fixed it for me:

1. sudo apt-get install libjack0
2. sudo apt-get install libjack-dev
3. pip install python-rtmidi

I am a noob and dont understand what they do, i just followed some forum instructions and it worked for me.

Shahroozgr, 2019-10-10:
thanks 
I have a question; Did you install and run the Piano LED Visualizer on noob?

ajayprk, 2019-10-10:
Oh no, I meant I'm a novice and don't understand coding or what these commands do, but they helped remove my error. I did this on RPi. Haven't been able to test the entire thing yet and patiently waiting for the LED strip to arrive by post today

Shahroozgr, 2019-10-10:
oh I got it :)
I am same as you and I have no idea about this error :(

ajayprk, 2019-10-11:
Did you get it to work? I had to reinstall from scratch due to a stupid mistake and I ran into the same issue once again. Apart from the commands above I also had to install libasound2-dev so in all

sudo apt-get install libjack0
sudo apt-get install libjack-dev
sudo apt-get install libasound2-dev
pip install python-rtmidi

Good luck

Shahroozgr, 2019-10-12:
not yet
I think I have to reinstall it again :(
I used raspbian buster full version; Is that ok? or I must use just raspbian buster lite?
please guide me

closed 2019-10-18 by Shahroozgr