onlaj/Piano-LED-Visualizer #21: Midi delay

glootkoopshire, 2019-10-11:
Hi, I got everything working, but there is a delay between midi and audio. Is there a way to fix?

glootkoopshire, 2019-10-11:
When you record midi and audio, then try to sync it together, it doesnt line up, the midi is faster is what i meant

onlaj, 2019-10-11:
Oh right, I'm aware of that. Unfortunately this is not a problem of my script but because of external library.
I created an issue but no response yet https://github.com/mido/mido/issues/205
If they will add an option to set absolute timing for midi messages I will update the script.

glootkoopshire, 2019-10-11:
Would a midi splitter work for this? I have a program on my pc that records midis well

glootkoopshire, 2019-10-11:
Ive found a temporary solution: it seems like after every 10 seconds in a track, the midi gets desynced by .4-.5 seconds. Idk if there’s anything you can do to fix that in your code or if its due to mido, but you can edit the midi using MidiEditor and adjust the midi every 10 secs to go forward by .4-.5 secs.

onlaj, 2019-10-11:
Nice catch. On the issue I linked I pasted the code for recording, I don't see anything that might cause the delay.
If the delay is consistent then it would be possible to write a script to fix that just after saving midi to file.

glootkoopshire, 2019-10-11:
Yeah it seems like .25 second delay per 5 seconds and .50 second delay per 10, i tested it on a 2 minute long track. for perfect alignment, i think the script would have to adjust every 5 seconds for the .25 second delay. Any idea on how to do this in the code? I am not proficient in Python at all lol. If a note is played at the 5 second interval, you would move the  notes in the next 5 second interval forward in time by .25 seconds. If a note is hit before the 5 second interval and lasts a little bit after, you would move it up by .25 seconds + the length of the note before it hit the 5 second interval. If there is just silence at the 5 second interval, you would extend it by .25 seconds.

glootkoopshire, 2019-11-12:
Hey, idk if you'll see this, but the easiest solution to this is to set the midi track from the RPi to about 114 bpm. I think the midi track records at 120bpm which is the standard bpm, through mido or whatever it is, but for some reason 120 is too fast to keep up with the playing? Just a heads up.

glootkoopshire, 2019-11-12:
Idk the exact number for a perfect alignment, but 113-115 has been working for me.

glootkoopshire, 2019-11-23:
Hi, I ended up buying a midi solutions quadra thru device that works really well for recording the midi (a lil bit expensive though lol). I was working on a script to fix the delay but couldnt get it to work, i think the code is too advanced for me. The one thing i was stuck on is what is the variable that holds the time of the midi? I had all the math worked out, I just couldnt figure out how to edit the midi files time, and at a certain time.

onlaj, 2019-11-24:
This is the line in which the note is saved
`saving.add_track("note_on", original_note, velocity, elapsed_time*1000) `       
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L1255

and this is how the method works:
```
def add_track(self, status, note, velocity, time):
        self.track.append(Message(status, note=int(note), velocity=int(velocity), time=int(time)))
```
https://github.com/onlaj/Piano-LED-Visualizer/blob/master/visualizer.py#L767

I wasn't sure how to understand [mido documentation](https://mido.readthedocs.io/en/latest/midi_files.html#about-the-time-attribute) but multiplying  time difference between notes in seconds by 1000 seemed to work. Changing it to different values will change the time delay between the notes, but it won't fix the drift between some of them (every 5 seconds as you mentioned).

omidfk, 2020-01-24:
hi onlaj
i have same problem, what should I do?

Kadnium, 2020-02-01:
@onlaj hi, I have this problem as well. I looked at the mido documentation and there's this line
"The time attribute is used in several different ways:
inside a track, it is delta time in ticks. This must be an integer."
And in your code it is milliseconds, is this the same as ticks? Have you tried converting the seconds to ticks using this function? 

```
mido.second2tick(second, ticks_per_beat, tempo)
Convert absolute time in seconds to ticks.

Returns absolute time in ticks for a chosen MIDI file time resolution (ticks per beat, also called PPQN or pulses per quarter note) and tempo (microseconds per beat).
```

Kadnium, 2020-02-11:
Anyone struggling with this, I have a relatively cheap solution, I bought a midi USB-USB connector that allows midi connection between two computers, it's this https://compasflamenco.com/midi-usbusb-p-4.html?language=en. I record the midi with pianoteq (any other program that records midi works) and use cantabile and loopbe to pass the midi input to the midi out and pianoteq.

onlaj, 2020-02-11:
So you can confirm it works, right?
I will add this to readme file and mention you.
The only thing I'm worried about is that this shop is always low on supply, they might run out of products quickly if I recommend it in tutorial.

Kadnium, 2020-02-11:
Yes mine just arrived today, I tested it and works perfectly. The device has 4 inputs and 4 outputs, just make sure to select the same port (from visualizer settings) that outputs midi from your computer.
Also the connector comes with two usb cables.

closed 2020-12-23 by onlaj