onlaj/Piano-LED-Visualizer #22: No playback port

Roloo38, 2019-10-16:
After i do all The things, in write The comand for The visualizer 
> sudo -E python3 visualizer.py
And i got an error “no playback port”
I want to Say that my Piano is ConnectedDrive through USB cablu to The raspberry, And The raspberry Find that port.
Any ideea Why?

onlaj, 2019-10-18:
Try to run /tests/printmidimessages.py" to check if RPi see messages from piano.

Roloo38, 2019-10-21:
I did this and I got another error: “ModuleNotFoundError: No module named ‘mido’” 
I forgot to install something or why I got this error?

onlaj, 2019-10-21:
If you installed everything from "requirements" file it should work.
Try without "3" in command
`sudo -E python visualizer.py`

Roloo38, 2019-10-21:
I tried without “3” and I got the same error. I will reinstall the req and I’m coming back if I doesn’t work. Thank you :)

Roloo38, 2019-10-21:
I reinstalled the requirements and I got another error: “ImportError: No module named webcolors 

https://m.imgur.com/gallery/cmngla9

Roloo38, 2019-10-21:
Ok, so, I ran “sudo -E python3.7 printmidimessages.py” and the RPI see my piano (“inport set to rtmidiout client:rtmidi output 132:0
Inport set to USB ProKeys 88:USB ProKeys 88 Midi 1 20:0
Inport set to USB ProKeys 88:USB ProKeys 88 Midi 1 20:1) so this means that the rpi see my piano.
If I ran “sudo -E Python visualizer.py” I got the “no playback port error”
Do you know why?

onlaj, 2019-10-21:
The most important is "inport", playback is needed when you want to play recorded midi.
What happens when you run printmidimesages.py and press piano keys? Does it print incoming messages?

Roloo38, 2019-10-21:
No, it doesn’t print anything on the terminal, it just show the message with the piano name.
If I play piano keys, nothing happens

onlaj, 2019-10-21:
So it seems your piano doesn't send any messages or RPi is not able to receive them. Does it work with PC or other equipment?

Roloo38, 2019-10-21:
Yes, I record with pc and usb cable and the piano send messages to pc

onlaj, 2019-10-21:
Well, I'm afraid that I can't help you. I had similar problem when I connected my piano through cheap midi to usb interface but it also didn't work with PC.

Roloo38, 2019-10-21:
So tell me, the connection between the piano and the rpi is made using only an usb cable, right?

onlaj, 2019-10-21:
In my case it is connected through midi to usb interface because my piano doesn't have USB.

Roloo38, 2019-10-21:
Ok, I think I have an midi to usb interface, I will try with it and let you know if it works for me :D 
Thanks for helping

closed 2020-01-07 by onlaj