onlaj/Piano-LED-Visualizer #25: Reverse light!!

Shahroozgr, 2019-10-22:
Hello
I use Raspberry pi 3b+ , Logic Studio sl-88 midi controller and usb connection.
After running the visualizer, I have two problems :
1- We have two leds  for each key but in my case only one led  works for each key. Is this correct or not?

2. When I press any key on my midi controller, the led does not work!! but after the finger is lifted off, the light goes on!!
Do you have any idea about this?
Tanks

onlaj, 2019-10-22:
1. yes
2. Run "tests/printmidimessages.py", play some notes and paste the console output here.

Shahroozgr, 2019-10-30:
Hello again
Unfortunately, I haven't been able to fix this bug yet.
As you said I tried to run the "tests/printmidimessages.py" but there was no reaction.
The MIDI keyboard was not working, LCD was white and LED was off.
I have a little knowledge in electronics, but I have completely followed all the instructions as you said.
when I configured automatic midi connection with “ connectall.rb “ command, I got this message:
Trackback (most recent call last):
         2: from /usr/local/bin/connectall.rb:6:in `<main>' 
         1: from /usr/local/bin/connectall.rb:6:in `each'
/usr/local/bin/connectall.rb:12:in  `block in <main>': undefined local variable or method `names' for main:Object (NameError)
But when I check the result with aconnect -l, I can see my MIDI keyboard:
client 0: 'System' [type=kernel]
0 'Timer '
1 'Announce '
client 14: 'Midi Through' [type=kernel]
0 'Midi Through Port-0'
client 20: 'SL STUDIO' [type=kernel,card=1]
0 'SL STUDIO MIDI 1'
1 'SL STUDIO MIDI 2'
Is it an error?! If it is, how to fix it? Or could that be the problem with my Midi controller?
Except the message that I don't know what it is, all the things was fine. 
Of course, I had an Error to install “ python-rtmidi “ but after installing “ jack0 “ and “ jack-dev “ I could install rtmidi. Everything installed but the LEDs not work properly!!!
Could you guide me please?

closed 2019-11-06 by Shahroozgr