onlaj/Piano-LED-Visualizer #26: Synthesia does not see the RPi Bluetooth

Hicershice, 2019-10-22:
Hello :)
I try everything since 2 weeks but I can't solve this problem.
I can connect my RPi to the laptop , but in Synthesia the RPi Bluetooth doesn't show up , I don't know what to do :(
You got some idea?

onlaj, 2019-10-22:
Did you replace .dll file and enable bt support in hidden synthesia settings?

Hicershice, 2019-10-22:
Yes, i replaced the file and activated "Midi.UseWinRTMidi"

If I try to connect laptop and RPi it says " connection failed - no usable services in this device" 
But they are paired all the time

onlaj, 2019-10-22:
You should connect to your RPi from PC, not the other way.

Hicershice, 2019-10-22:
Oh I did not know that there's a difference between connecting "from device a to device b"
 to connecting "from device b to device a" 😅

onlaj, 2019-10-22:
It works without screen but you can't change any settings.

Hicershice, 2019-10-23:
Thanks!

Now the Bluetooth Input works but in the other console i Run visualizer.py and it says "outport set to RPi Bluetooth:RPi Bluetooth 128:0"
Shouldnt the Bluetooth bei the Input and the LED String ne the Output?

closed 2019-10-23 by Hicershice