onlaj/Piano-LED-Visualizer #27: LEDs dont respond to keyboard

Hicershice, 2019-10-23:
Im very Close to get Things working now.
But somehow The LEDs dont respond to Signals from The Keyboard.

The Bluetooth server recognizes the keyboard Signals , but the LEDs don't show any reactions.

When I run visualizer.py it says:
Outport set to RPi Bluetooth:RPi Bluetooth 128:0

And when I run aconnect -l it says:
client 0: 'System [type=Kernel]
      0 'Timer                    '
      1 'Announce             '
client 14: 'Midi Through' [type=Kernel]
      0 'Midi Through' Port-0'
client 128: 'RPi Bluetooth' [type=user,pid=1283]
      0 'RPi Bluetooth'

Hicershice, 2019-10-24:
I realised I had not activated light support on Synthesia ..
But now I don't know which channel to use for light support. The lights don't react to any channel.

When I enter the settings for light support, the Bluetooth server gets hundreds of incoming signals . I don't understand it

Hicershice, 2019-10-24:
Do I have to start sudo -E python visualizer.py ?
Or does it start on boot?

Hicershice, 2019-10-24:
If I try "sudo python tests/printmidimessages.py" I get a NameError : name 'Inport' is not defined
And if I debug the script there is a ModuleNotFoundError: No Module named mido

Hicershice, 2019-10-24:
And when I run visualizer.py and choose 'la campanrlla' in 'play midi' on the screen , the LEDs don't light up. 
But when I click on 'led animations' and choose something, the LEDs light up. 

Please help me

Hicershice, 2019-10-24:
Okay, I have found the solution . So much work!

I had to use the screen to change the 'input' (why?) To RPi Bluetooth, it wasn't set to it as it seems

closed 2019-10-24 by Hicershice

Dontbemadpls, 2019-11-27:
Did you find a solution to this?