onlaj/Piano-LED-Visualizer #36: delay and skipped note on rpi 3b+

jihedhajali, 2019-11-13:
Hi ,

Thank you for your effort ! i have been looking for a project like this for a while now :)
Unfortunately i am experiencing some difficulties when using visualizer on a rpi 3b+ (some how no one using rpi zero is concerned by this ... ) . when i touch a key on the keyboard there is a 1-2 second delay ! and when i do it fast , it skips many note ! 

Thank you .

onlaj, 2019-11-13:
Hello.
I don't have 3b+ to test it with my own setup, I can only suggest you to profile the script with [cProfile](https://docs.python.org/2/library/profile.html) to check which parts of codes takes so much time to process.

jihedhajali, 2019-11-13:
hi , looks good to me ! I tried running lightkeys.py from the tests folder and i noticed that i constantly get an error message "midiInAlsa:alsamidiHandler: unknown midi input error" "ressource temporarily unavailable"

konradqwerty, 2019-11-23:
Hi, did you find any solution? It also happens to me

jihedhajali, 2019-11-24:
hi , no solution yet . i found that when i connect the rpi to synthesia throw Bt there is no lag anymore , it's only when i select my piano as an input/output !! ( i have a yamaha p45 btw )

Sinope81, 2020-01-04:
Hi, I have same problem: when i push a key on Yamaha P60 than the LED strip light up with 0.5-1.0 sec delay or skip it.
I use Raspberry Pi 3B+, MIDI-USB cable, WS2813 LED strip (and Waveshare HAT, but it doesn’t work well, too). I tried with a silicon roll up piano (USB), but it have delay, too.

konradqwerty, 2020-01-04:
I couldn't make it work, but I found MuseScore PianoTutor, which worked for me after some trials.

Kadnium, 2020-01-17:
Hi, I have this same problem with model 3b, the leds are delayed when a key is pressed.

closed 2020-02-22 by onlaj