onlaj/Piano-LED-Visualizer #38: no playback port & inport set to RtMidiOut

Dontbemadpls, 2019-11-21:
I'm running the code on a RPi 3 v1.2, the piano has midi capabilities (I've accomplished similar results with an arduino + neopixel) so I know it can send proper midi data through usb. 

I've installed the prerequisites and set DisablePlugins= pnat and enabled SPI, I've followed all the instructions and even doubled down on them to make sure the files created were all correct and there weren't any typos. 

Here is the issue: 

The LCD is jamming out and flickers like mad
When running visualizer in sudo -E the keys don't light up
When running visualizer in sudo (no -E) they light up but with a noticable delay until an eventual failure
upon running sudo -E visualizer python visualizer.py i get following messages:
`Inport set to ARIUD:ARIUS MIDI 1 20:0
Inport set to RtMidiOut Client:RtMidi output 131:0
Inport set to RtMidiOut Client:RtMidi output 134:0
Inport set to RtMidiOut Client:RtMidi output 135:0
Inport set to RtMidiOut Client:RtMidi output 136:0
Inport set to RtMidiOut Client:RtMidi output 141:0
Inport set to RtMidiOut Client:RtMidi output 147:0
Inport set to RtMidiOut Client:RtMidi output 148:0
Inport set to RtMidiOut Client:RtMidi output 149:0
Inport set to RtMidiOut Client:RtMidi output 150:0
playport set to ARIUS:ARIUS MIDI 1 20:0
no playback port`

I'm using a WS21813 LED strip, and wired it up just like normal however with the backup data cable put to the same GND on the RPi as the GND coming out of the strip. I am confident in my wiring, I think there may be something wrong with Pulseaudio being run in the wrong environment, I briefly got an error message (when running from terminal) stating 

`xdg_runtime_dir (/run/user/1000) is not owned by us`

But it disappeared after reverting to the desktop environment. I've made sure SPI is enabled at all times, I have no idea what to do now and any help will be greatly appreciated!

onlaj, 2019-11-21:
Hello.
Did you disabled audio output as described here? https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/

> The audio output must be deactivated. For this we edit the file
> `sudo nano /etc/modprobe.d/snd-blacklist.conf`
> Here we add the following line:
> 
> `blacklist snd_bcm2835`
> Then the file is saved by pressing CTRL + O and CTRL + X closes the editor.

Dontbemadpls, 2019-11-21:
Hey Onlaj, 
Yes I did disable audio output as described, thank you for the quick response! I just double checked and the blacklist is still active.

Dontbemadpls, 2019-11-23:
Update: For anyone having the same issues with flickering lcd, I apparently was running two instances of visualizer.py ontop of eachother (some stupidity with my boot settings) which resulted in the described effect. 

I reinstalled everything and that seemed to do it for the rest of the issues.

closed 2019-11-23 by Dontbemadpls