onlaj/Piano-LED-Visualizer #39: Length of LED strip too short?

baksteen-13, 2019-11-22:
I would like to build this for my piano but before ordering i noticed that the led strips linked on the github that the one on Amazon isn't the minimum length of 1.5m. so does that mean the only realistic option is buying the "2m 144 IP30" from the aliexpress link?

-[https://www.amazon.com/BTF-LIGHTING-WS2812B-Individual-Addressable-Non-waterproof/dp/B01CDTEJR0/ref=as_li_ss_tl?keywords=WS2812B&qid=1563793619&s=gateway&sr=8-3&linkCode=sl1&tag=onlaj-20&linkId=1ec35f385e9049b6ecfb25619d55c562&language=en_US](url)

-[https://nl.aliexpress.com/item/32832420003.html?aff_platform=link-c-tool&cpt=1574455768121&sk=dFyC7NO&aff_trace_key=03be746d74954420917d5fcbb464cbf3-1574455768121-07753-dFyC7NO&terminal_id=969ed013c1cb461fb6dc0fc56b144071](url)

onlaj, 2019-11-23:
Hello.
You can buy 2 pieces of 1 meter long and solder them together.

baksteen-13, 2019-11-23:
Thanks for the quick response i will take a look.

closed 2019-11-23 by baksteen-13