onlaj/Piano-LED-Visualizer #40: 61 keys

fischer6464, 2019-11-26:
Thank you for providing this guide.

Is it possible to edit the code to use with a 61 keys keyboard?
How ?

onlaj, 2019-11-26:
Hey, find in visualizer.py file number "176" (2 leds for every 88 piano keys), it should occur 4 times, and change it to "122". It should work.

fischer6464, 2019-11-28:
Thnx

fischer6464, 2019-11-28:
One more question, can I edit the number of skipped leds?

fischer6464, 2019-11-28:
[image]

onlaj, 2019-11-28:
This part of code (first 6 lines) is for fixing key spacing. So if the led lights up in the wrong place you can fix it by changing note_offset in specific keys range.

fischer6464, 2019-11-28:
Great 
I made it work btw. Just need to edit the py file to fix the led numbers to 122.

You been a great help Onlaj 

Thank you

closed 2020-01-07 by onlaj