onlaj/Piano-LED-Visualizer #41: Was working perfectly! Now doesn't recognize keyboard

Babeeplz, 2019-11-27:
Hey, appreciate your work! Quick question:

I had this device built for me and it was working perfectly, synccing to my keyboard no problem for eg. CASIO on the input list.  Then suddenly next day, one light on the LED strip perma stayed on, and soon after it became unresponsive.

Now, it reads all keyboards as USB DEVICE, and doesn't activate the lights anymore (but the lights still work when tested using LED ANIMATIONS.

It's definitely not hardware and have tried multiple different cables keyboards and unplugging, it seems to be a driver issue.  Any idea how to get it to read my keyboard as CASIO like before? As i suspect this is the problem.  Any help/ fix instructions greatly appreciated!!!

[image]
[image]

onlaj, 2019-11-28:
Hey.
There are so many things that could go wrong that my only advice for now is to reinstall the whole Raspbian system.
If it will happen again then we can try to find some solution. Maybe reinstalling/reseting some linux service would do the work.
btw. what 3d printer and filament did you use? It seems to be much higher quality than mine :D

Babeeplz, 2019-11-28:
Hey, thanks for your response!
I had a really pro engineer build it for me so I have no idea how he did it!
Is it possible to give the instructions on how to 'reinstall the whole Raspbian system'?
Thanks so much!

Dilanka-lakshan, 2019-11-29:
How did you find a case for the both waveshare and for raspberry pi

Dilanka-lakshan, 2019-11-29:
Can u leave a link to buy it please ...

onlaj, 2019-11-30:
Read instructions, 3d file is attached in repository, you have to print it on 3d printer.

onlaj, 2019-12-02:
@Babeeplz 

> Could you please give me instructions on how to 'reinstall the whole Raspbian system'?

Remove sd card from RPi, put it to card reader in your pc and format it. Then install Raspbian and repeat all steps mentioned in tutorial.

Babeeplz, 2019-12-10:
Thankyou onlaj, and thankyou for your work it's amazing!

closed 2019-12-10 by Babeeplz

Dilanka-lakshan, 2019-12-12:
How much A does it need to light up DC5V WS2812B 144 strip to maximum
brightness? Is 6A is enough for that .

onlaj, 2019-12-12:
Each diode takes up to 0.06A, so you need 9A.

Dilanka-lakshan, 2019-12-12:
Ok thank you