onlaj/Piano-LED-Visualizer #42: Wiring the LCD hat to breadboard

Dontbemadpls, 2019-11-28:
Hi Onlaj, how did you wire everything around the first time around? I'm having some difficulty with the breadboard connections. 

I'm especially interested in how the waveshare 1.44in lcd hat should be wired up, not so much the rpi to the breadboard (which pretty straightforward)

onlaj, 2019-11-30:
> I'm especially interested in how the waveshare 1.44in lcd hat should be wired up, not so much the rpi to the breadboard (which pretty straightforward)

You need RPi with presoldered pins, then you just put Waveshare screen over them.

Dontbemadpls, 2019-11-30:
How would you wire it all together using a breadboard?

onlaj, 2019-11-30:
You would have to read Waveshare documentation to check which pins you need to connect.

Dontbemadpls, 2019-11-30:
I'm quite new to RPI, do you recall what you used?

onlaj, 2019-11-30:
I didn't use anything, I just put screen on pins, nothing else.

Dontbemadpls, 2019-12-01:
I have found this following description: 

RESX: reset. Set it to 0 while powering the module. Usually set it to 1
IM2: data communication mode pin, which define usage of SPI
CSX: chip selection control pin. If CS=0 – the chip is selected
D/CX: data/command control pin, if DC=0 – command is written. And if DC=1 – data are written
SDA: transmitted RGB data
SCL: SPI clock

As far as I can see, this means I need to connect the lcd to:
rpi pin 01: 3.3v DC
GPIO 02: SDA1
GPIO 11: SPI_CLK
ground? RXD0? 

I'm unsure how to continue, any help would be greatly appreciated. Thank you in advance.

Roloo38, 2020-03-12:
@Dontbemadpls Did you figured it out? If you didnt, i think i can help you:)

Dontbemadpls, 2020-03-12:
Hey @Roloo38 ! Any help would be appreciated, I'm still in the same boat since last I wrote.

Roloo38, 2020-03-12:
Well, do you have facebook or instagram? We can talk there :)

Dontbemadpls, 2020-03-12:
Instagram would be fine, my usernames _.e.d.t._, feel free to hit me up any time but I only have the time tomorrow since it's quite late here now

Roloo38, 2020-03-12:
I cant find you.. when you have time, you can add me by the link

https://instagram.com/raulobretin?igshid=ym6f3cllnvfd

closed 2020-12-23 by onlaj