onlaj/Piano-LED-Visualizer #43: Raspberry pi zero wh is not available

Dilanka-lakshan, 2019-11-29:
Raspberry pi zero is not available in the Ali Express  link you mentioned and its out of stock. I cant buy from amazon cuz there is some problem in there. So can u update a another seller with this item in Ali Express 
[image]

Dilanka-lakshan, 2019-11-29:
[image]

onlaj, 2019-11-29:
I updated the links.

closed 2019-11-29 by onlaj

Dilanka-lakshan, 2019-11-30:
Can u please mention a cheap price from an another seller its 36 $. I am
from srilanka and. Its a big price when i convert it in to my currency.  It
like 36 × 180 = 6480 LKR

Dilanka-lakshan, 2019-11-30:
Can i use 10v power supply to light up 100% 172 lights

onlaj, 2019-11-30:
I didn't find it cheaper, maybe it is out of stock because of Black Friday, I don't know.

> Can i use 10v power supply to light up 100% 172 lights
You need 5V (Volt) and 11A (Ampere) to light them all @100% and white color.

Dilanka-lakshan, 2019-11-30:
Thank you for your response. I ll see u soon.

See you later....