onlaj/Piano-LED-Visualizer #44: DC5V WS2812B is not working after plugging to Dc5v

Dilanka-lakshan, 2019-12-06:
After connecting to Dc5v its only lighted up the first led bulb and it was green and others were not. Why is that. Does it need a controller or something to work the whole strip

onlaj, 2019-12-06:
Yes, the led strip needs to receive data from RPi or some other controller.

Dilanka-lakshan, 2019-12-08:
how can i assure that the strip working correctly without a controller. still i dont have a controller. how to test that is working? i ordered form the link you mentioned in your site. i want to verify it working correctly. i got the order.

Mihairu023, 2020-02-04:
Depends. If you've got a Raspberry Pi then connect everything using the instruction and [this picture](https://tutorials-raspberrypi.de/wp-content/uploads/Raspberry-Pi-WS2812-Steckplatine-600x361.png), prepare your RPi and test the WS2812B by launching visualizer.py. 
If not, get a Raspberry Pi (Zero WH is preferrable).
Easier way to just test the LED strip is to buy any WS2812B controller, [Google](https://www.google.com.ua/search?sxsrf=ACYBGNQdzzQjWbWT4ETvb6KvZmZ8LNsC-w%3A1580858157990&ei=Lfs5Xp2CPLH4qwHH_qe4Dg&q=ws2812b+controller&oq=ws2812b+controller&gs_l=psy-ab.3..0i203l7j0j0i203j0i22i30.1163.2421..2534...0.1..0.93.1061.12......0....1..gws-wiz.......0i71j35i39j0i67.zA3_82CBRVk&ved=0ahUKEwjdzMnHg7nnAhUx_CoKHUf_CecQ4dUDCAs&uact=5) and [AliExpress ](https://btf-lighting.ru.aliexpress.com/store/group/Pixels-Controller/1051119_508208217.html?spm=a2g0v.12010612.0.0.6a5fdd367w1tK4) offer a wide variety of those (I used [this](https://aliexpress.ru/item/32789402969.html?spm=a2g0s.9042311.0.0.1ec633edI7jGNF&dp=e7ac4027f279380d631eba8d55a0d9aa&dp=e7ac4027f279380d631eba8d55a0d9aa&af=886937&af=886937&cv=47843&cv=47843&afref=&afref=&mall_affr=pr3&mall_affr=pr3&aff_request_id=c8e3b9e3e869429d87ebc1dd00ab02b0-1580858872447-07213-VnYZvQVf&aff_platform=aaf&cpt=1580858872447&sk=VnYZvQVf&aff_trace_key=c8e3b9e3e869429d87ebc1dd00ab02b0-1580858872447-07213-VnYZvQVf&terminal_id=d64c2d2a9fb643e6b7aefa1d9ec5eb0f) because of WiFi control support, this link is not affiliate). Depending on results of the test you can proceed to acquire an RPi and other required items (whichever you need) or repair/change your WS2812B strip if it's broken. 
Anyway, I guess the photo of your strip's connection will be welcomed here as it may make it easier to help you.

The-Torrent, 2020-02-08:
> how can i assure that the strip working correctly without a controller. still i dont have a controller. how to test that is working? i ordered form the link you mentioned in your site. i want to verify it working correctly. i got the order.

if you quickly tap the data wire on ground led's will light up.

The-Torrent, 2020-02-08:
> Depends. If you've got a Raspberry Pi then connect everything using the instruction and [this [...]

that photo you got in 'this picture', is that how led strip is wired up for this project?

Mihairu023, 2020-02-08:
> that photo you got in 'this picture', is that how led strip is wired up for this project?

Yes. Not being skillful with wiring (and schemes like that depicted on the picture), I've been searching for more detailed steps to set things up. The trick is to put two ground jumper wires into the same "-" socket of the [connector](https://tutorials-raspberrypi.de/wp-content/uploads/DC-Steckeradapter-CCTV-180x165.jpg). 
Not sure of this one but putting wires themselves is probably better than putting metal pins there as it may get harder to fix metal pins securely (I have the "pins" version right now, guess that's the reason of leds randomly lighting up).
Overall instruction: [(no, you need 144 leds/m, not 60)](https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/)
[And a picture of how did I connect it all.](https://imgur.com/a/B6XEXAu)

The-Torrent, 2020-02-09:
> > that photo you got in 'this picture', is that how led strip is wired up for this project?
> 
> Yes. Not being skillful with wiring (and schemes like that depicted on the picture), I've been searching for more detailed steps to set things up. The trick is to put two ground jumper wires into the same "-" socket of the [connector](https://tutorials-raspberrypi.de/wp-content/uploads/DC-Steckeradapter-CCTV-180x165.jpg).
> Not sure of this one but putting wires themselves is probably better than putting metal pins there as it may get harder to fix metal pins securely (I have the "pins" version right now, guess that's the reason of leds randomly lighting up).
> Overall instruction: [(no, you need 144 leds/m, not 60)](https://tutorials-raspberrypi.com/connect-control-raspberry-pi-ws2812-rgb-led-strips/)
> [And a picture of how did I connect those ground wires.](https://imgur.com/a/B6XEXAu)

the more issue i'm having is figuring out how the power wires go into the raspberry pi zero itself.

also not 60/meter, but 72 so that its 1led per meter im wondering if that would work.

I'll probably attempt a build in march some time when i have a evening free or so i could test it out.

Mihairu023, 2020-02-09:
> the more issue i'm having is figuring out how the power wires go into the raspberry pi zero itself.

Powering the RPi seems to be the easiest part - power bank does it good. 
The harder one is to fix data wire and one of the ground wires to RPi - you either need to **solder** them to corresponding pins **or** you need to **remove insulation** at the "RPi side" (over 1 cm or 0.4 inches) and **wrap** the bare wires over the pins so they don't touch the adjacent pins. In the second case heat shrink bands will help you a lot (they're cheap, AFAIC). [That's how Onlaj did it.](https://i.imgur.com/7KhwM7r.jpg)
If you're fine with white color only (or you know Python a little and can change settings manually) you can skip the soldering/wrapping part and just connect everything with jumper wires. But as for me, that's rather temporary solution as you won't be able to attach the LCD hat that lets you configure the lighting how ever you want.
Anyway, RPi needs only two wires - ground (from power supply to right pin, third from top) and data (from LED strip to right pin, sixth from top). "That picture" at one of the upper comments may explain it better.

> also not 60/meter, but 72 so that its 1led per meter im wondering if that would work.

_Please no_. Already failed because of similar reason - now I occasionally use that "not-so-dense" led strip for house lighting. Long story short - **visualizer won't work** properly with led strips other than 144 leds/meter, and also "144 strip" is **the only one that can cover every single piano key** - otherwise a lot of keys will be skipped.
So please get a 144 leds per meter led strip.

> I'll probably attempt a build in march some time when i have a evening free or so i could test it out.

Good luck!

P.S. It is said that _you can discard the external power supply_ as well - RPi's power can light up over 20 leds simultaneously. But that's suitable **for testing purposes only**: you won't be able to play piano pieces that are too dynamic/difficult and led animations will go to waste too.

The-Torrent, 2020-02-09:
> Powering the RPi seems to be the easiest part - power bank does it good.
> The harder one is to fix data wire and one of the ground wires to RPi - you either need to **solder** them to corresponding [...]

Fixing data wires won’t be too hard imo for me anyway I got loads of shrink wrap no biggie.

I think I know how to make it all now will attempt in March.

Also 144/meter is 2 LED’s per key, not 1. I don’t see advantage of using 2 over one as it’s just harder to power. Much harder. Have you tried 72/meter at all? (72x1.22 = 88, 88 keys on piano). This is the strip density I have used for all my other visualisers I have made with arduino and esp methods.

Just a quick few questions for you as you’ve built it, is ws2812b latching issue around? As I transitioned to apa102b for the clock pin due to the latching issue as well as better colour etc, but me using ws2812b would be much easier than recoding it to work with apa102 if the latching issue does not exist.

Do you use all LED’s at 50% power or max?

On pre-programmed modes, I’m not sure how many there are, for each key pressed does 1 led turn on or 2? As density wise there is 2 per key. (144 x 0.22 = 172, 172/88(keys in piano) = 2)

Thanks

Mihairu023, 2020-02-09:
That's great then.

As for 72/meter strip, can't tell much because I didn't try it indeed. Everyone who tried using it will be welcomed here
Guess it all depends on the piano used for the project though, as Onlaj suggested that 176 leds are needed (at least that's default value specified in code) while my Yamaha P-45 only fits 174 of them, and 172 is your value. Seems like a question of guessing.

Don't know this one too, alas. Judging by what I've googled apa102 is better than ws2812b but is less known and uses the whole different protocol.
<whining>
Still stuck with first led always glowing in white/light blue color, sometimes long delays, sometimes only 6 first leds half-working/half-flickering... in other words there was everything between "working really smoothly" and "nothing but that uncontrollable first led works". Yet all the time I've just made sure that pins of jumper wires touch RPi's pins without any additional fixing. What makes me wonder is why did it managed to work anyway (although it happened only once, yet still video is available) and how do I _actually_ wrap those stranded and quite brittle wires around the pins; last time I tried to do it, wires either got out of "the single wire" shape sometimes resulting in embracing all the adjacent pins too or they began to crumble when I tried to make them embrace the required pin only. Bare enthusiasm, lots of tries and little to no knowledge of wiring and stuff - that's how it started and that's how it keeps going on for me. Didn't even know about WS2812B, RPi and stuff before seeing this manual. Looks like I still don't know a lot of related things.
</whining>
Sorry for that. Actually I didn't even understood what is the latching issue you've mentioned. No offence.

Regarding the LED power - it's 50%, I've ordered the "5v 6a" power supply as suggested and the strip still shines brightly enough. 

One led per key. The code contains specific offset function - because without it even with 144/meter strip being used, wrong leds (with inaccuracy of 1-2 leds) are getting lit in the right half of piano... _presumably_, as I didn't yet try to remove/edit that function.
BTW, there's an option to let 3 leds per key be turned on instead thanks to [Dec's commit](https://github.com/onlaj/Piano-LED-Visualizer/commit/74ef5376033c7274060b1c3279ecd3a6c42ebac2).