onlaj/Piano-LED-Visualizer #49: sometimes my raspi stops on this screen

hyuny223, 2019-12-13:
sometimes my raspi stops on this screen
Is there a way to solve this problem?
[image]

onlaj, 2019-12-13:
Hey.
This is a screensaver, if you have keyboard connected it will exit this mode when you start playing. You can also hold KEY2 button.

hyuny223, 2019-12-13:
I pressed KEY2 button but i didn't work.
So I always should turn off the device on the first attempt.

onlaj, 2019-12-13:
Hold KEY2, not just press it.

hyuny223, 2019-12-13:
When my device enter screen saver mode, it all stops. Time also does not go. Holding key didn't work... :(

onlaj, 2019-12-13:
Oh, so it maybe crashes. Can you show me console output after the crash?

hyuny223, 2019-12-13:
Sorry for troubling you. 
The picture I uploaded is the console output after the crash.
When it entered screensaver mode, screen got stucked like that.
For operation, I should unplug power cable.

onlaj, 2019-12-13:
By console I meant the window in which you typed "sudo -E python visualizer.py". It should show information why it crashed.

hyuny223, 2019-12-13:
ah! I misunderstood!
[image]

onlaj, 2019-12-13:
You didn't even start the script.
Run it from the correct directory so you can see working menu on RPi screen and wait until screensaver starts. Then check console for errors.

hyuny223, 2019-12-13:
I am not an English speaker, I am a coding beginner. So I don't know whether I exactly did what you said.
[image]

onlaj, 2019-12-13:
Ok, so it crashes because you have no piano attached. I will fix that in next update. For now you can avoid getting this error by connecting your piano to RPi.

hyuny223, 2019-12-13:
ah! thanks for your reply. I got it.
After solving this problem, what should I do to update?
type "sudo apt-get update"???

onlaj, 2019-12-13:
https://help.github.com/en/github/using-git/getting-changes-from-a-remote-repository

hyuny223, 2019-12-13:
Then I need to clone your git!

closed 2020-12-23 by onlaj