onlaj/Piano-LED-Visualizer #52: Question regarding the Display

Hicershice, 2019-12-16:
i build everything like you said in the tutorial and thanks for that!

now i want to buy a raspberry pi 3 , will it work with that raspberry? and what about the display ? do i have to buy another and will another display work with the scripts?

onlaj, 2019-12-16:
Hello.
Yes, it will work with RPi 3, but some users reported problems with latency. I don't have R3 so I can't test it myself. 
Anyway, RPi 3 is an overkill for such a task and obviously it won't fit into my custom case.
My script will only work with screen mentioned in tutorial.

Hicershice, 2019-12-16:
It happened to me that the LEDs dont respond very fast and some LEDs dont respond at all when i am playing fast. i thought that i could change that fact with switching to RPi3.

onlaj, 2019-12-16:
I don't have such problems with my RPi Zero.

> Maybe its possible to skip the bluetooth midi connection and connect the piano via Midi cable directly with the RPi? this could kill the latency or am i wrong?
This is how it suppose to work. I have connected my piano directly to RPi, no bluetooth. BT is only for Synthesia.

Hicershice, 2019-12-16:
ah okay, but when i connect the piano via cable i cant connect my mouse to the rpi to start the scripts ^^

onlaj, 2019-12-16:
Set up SSH or VNC connection. You can also buy USB HUB.

Hicershice, 2019-12-16:
i will try, thank you

Hicershice, 2019-12-20:
i connected via SSH to the raspberry and connected the piano without the bluetooth server to the raspberry, but now there is no option for the visualizer.py to put the input to usb.
you say BT is only for synthesia.. but BT was the input for MIDI for the script or wasnt it ?

Hicershice, 2019-12-20:
NVM i fixed it ^^

closed 2020-01-07 by onlaj