onlaj/Piano-LED-Visualizer #55: LEDs don't respond to key press

hyuny223, 2019-12-28:
I followed your guide and operated my raspi.
But when I pressed key, LEDs didn't work(but led animation wokred).
So I formatted my sd card and restarted installation. But the result was same.
what's the problem?? any solution?

hyuny223, 2019-12-28:
my piano has a midi port like this.
is it related to this problem?
[image]

JAKJUS7, 2020-01-02:
Probably. I am not sure, but I think that is like a midi input rather than the output you need for it.

onlaj, 2020-01-02:
Does your piano work when you connect it to your PC?

hyuny223, 2020-01-04:
yes. connecting piano to my PC, it worked well.
but when connecting to raspberrypi, it didn't work.
absolutely, i know it was connected because the screen savor worked.

hyuny223, 2020-01-04:
my piano name is "MPS-120" in kurzweil.

hyuny223, 2020-01-04:
Is "pip install mido" different from "MIDI Objects for Python" you uploaded in your Youtube?

devucsb, 2020-12-07:
Having this same issue. Animations working fine, but not responding to keys being pressed. My PC is picking up MIDI from the piano so should be good to send MIDI to RPI. My keyboard has the same MIDI port (printer cable)

HansiJung, 2021-01-07:
I have the same problem.

LED animation is working.
But no led, when i press a piano key.
What can I change in the set up?

I tryed the USB from the Yamaha piano with my Android tab. 
Piano USB B --> Otg Android tab and the App MIDI keyboard.
If I press a key on the Piano, than I see it in the APP. So I think, that the Piano is sending something.

Can you help.

My daughter is loving that LED, but now I am a looser, because it is not working ;-(

devucsb, 2021-01-08:
> I have the same problem.
>
> LED animation is working.
> But no led, when i press a piano key.
> What can I change in the set up?
>
> I tryed the USB from the Yamaha piano with my Android tab.
> Piano USB B --> [...]

go to port settings and refresh ports. After that select your digital piano as the input. With any luck you should be good to go. This tends to happen if the Pi boots up without the piano on so it doesn’t detect the piano while the pi is booting. Let me know if that worked!

HansiJung, 2021-01-08:
Hi,
after refreshing ports, I choose Input.
There are 3 choices
Midi through: midi through Port-0 14:0
ATRRTMIDIout Client: RTMIDI output 132:0
rtpmidi raspberrypi:Network 128:0

I tried all of them, but it did not work.

HansiJung, 2021-01-17:
I solved it.
The problem was, that I used the wrong USB port on Raspberry PI Zero.
I did not know, that one is only for power and the other one is the USB port!

Second improvemet was, that i connected the +5V to GIPO 2/4, so do not need an extra 5V USB powersupply.

Thank you very much for the project.

closed 2021-08-16 by onlaj