onlaj/Piano-LED-Visualizer #56: Waveshare LCD glitching

bradliegek3, 2019-12-31:
Hi Onlaj,

Thanks for the project. Almost everything works.

I only have an issue with the LCD screen. I am using the same waveshare screen.
When the rpi is booting the screen is white (nothing weird). After a while the menu shows. Now when I wait a couple seconds, the screen will be very glitchy and jumpy. Sometimes flashing green and sometimes switching to white again. When it has to white it will not come back to the normal menu untill I restart the rpi.

Sometimes the menu does work a couple minutes, but when I scroll though the menu, I see that key one sometimes needs to be pressed 2 to 6 times to work.

Do you have any idea how to fix this? The screen is brand new.

Thank you in advance and a happy 2020!

onlaj, 2019-12-31:
Hello.
Do you have your led strip connected to pins? Make sure the wires doesn't touch unwanted pins.
Also make sure than screen is properly connected, you have to use some force to put pins deeply in holes.

bradliegek3, 2019-12-31:
thanks for the quick response!

I am still waiting for the leds so they are not connected. the screen is pushed as far as possible on the pins. It leaves a little gap though between the pins and the screen connector.

onlaj, 2019-12-31:
What happens if you launch RPI without running Visualizer,py? Does the screen goes crazy too?

bradliegek3, 2019-12-31:
I will try later today as I need to install everything again since my sd card files gone corrupt this morning.

bradliegek3, 2019-12-31:
After starting the rpi without running visualizer, the screen is white.
Now when I start the visualizer, everything works fine.

So is the problem caused by automatically starting visualizer on startup?

onlaj, 2019-12-31:
Maybe somehow the script booted twice or more?

bradliegek3, 2019-12-31:
No idea how. I just add
 sudo -E python /home/pi/Piano-LED-Visualizer/visualizer.py &
to /etc/profile

bradliegek3, 2020-01-01:
Or is that the wrong way to automatically boot the visualizer?

bradliegek3, 2020-01-09:
Also, when the LCD is glitching, the LED's do not work properly. When I boot the rpi without visualizer and then manually type: 
cd Piano-LED-Visualizer
sudo -E python visualizer.py
only then the screen and the LED's react nicely.

How do I run visualizer.py on startup properly?

Type sudo nano /etc/profile
Scroll to the bottom, and then??

bradliegek3, 2020-01-14:
> Maybe somehow the script booted twice or more?
I have tried the following:

-add "sudo -E python /home/pi/Piano-LED-Visualizer/visualizer.py &" to "/etc/profile". -> glitching lcd and slow led's

-add "sudo python /home/pi/Piano-LED-Visualizer/visualizer.py &" to "/etc/profile". -> glitching lcd and slow led's

-add "sudo -E python /home/pi/Piano-LED-Visualizer/visualizer.py" to "/etc/profile". -> RPi stopped not booting (black screen).

-add "cd Piano-LED-visualizer" and "sudo -E python visualizer.py" to "/etc/profile". -> glitching lcd and slow led's.

Right now I am completely out of options. This is the second RPi zero I have tried it on. The third sd card and I have re installed everything over a dozen times.

BendixMa, 2020-01-25:
Yea, I had kind of the same issue. For me I fixed it by booting into CLI (terminal) instead of desktop and disabling ssh. Then I rebooted the pi and enabled ssh again (if you need it). Now its working fine. It must because the script is started multiple times (or the old script isn't stopped properly), because when I ssh into the pi the script is started again and the glitches also start to appear again and only a reboot solves the problem.

bradliegek3, 2020-01-26:
That actually worked! Only problem then is that I can't use VNC when the RPi starts in console. So that gives a problem with connecting to bluetooth.
Do you use the bluetooth, if so, how?

BendixMa, 2020-01-29:
No I'm not using Bluetooth but do you really need the desktop GUI for that? I added some code in my last pull request that tries to ensure that there is only instance of the script running at the same time. So you could pull the new changes and enable desktop GUI again and see if it works now without the LCD glitching. Alternatively to `VNC` you could also try `xrdp` as a remote desktop which works for me even in console mode.

bradliegek3, 2020-02-27:
Or it doesn't work, or I do not know how to pull the new changes :p.
Can you give me the instructions on how to get the latest version of the files on my RPi?

vzoltan, 2020-02-27:
> ....because when I ssh into the pi the script is started again and the glitches also start to appear again and only a reboot solves the problem....

this is why i made a service, and visualizer runs as service, not when pi user logs in.

bradliegek3, 2020-02-27:
Can you explain what I need to do?

vzoltan, 2020-02-27:
Make sure visualizer runs only once.

Best way to remove from 
/Etc/profile

And then do these steps

sudo nano  /lib/systemd/system/pianoled.service
Add this to the file:

```
[Unit]
Description=PianoLED
After=multi-user.target
Conflicts=getty@tty1.service
[Service]
Type=simple
ExecStart=sudo /bin/sh -c 'HOME=/home/pi /usr/bin/python /home/pi/Piano-LED-Visualizer/visualizer.py'
StandardInput=tty-force
[Install]
WantedBy=multi-user.target
```
 Save it and then run these

sudo systemctl daemon-reload

sudo systemctl enable pianoled.service

sudo systemctl status pianoled.service  

if needed start:

sudo systemctl start pianoled.service

after reboot it will start ;)

bradliegek3, 2020-02-27:
Well, this was by far the best and easiest solution. Thanks vzoltan!

closed 2020-02-27 by bradliegek3