onlaj/Piano-LED-Visualizer #57: Automatic bluetooth connection

bradliegek3, 2019-12-31:
Hi Onlaj,

Thanks for your project!

I have a question about the bluetooth function.

Is the rpi supposed to always send a rpi bluetooth signal so I can connect it to my laptop? Because mine doesn't. When I start the rpi, piano and laptop; I first have to acces the pi with my laptop to excecute the command sudo btmidi-server -v -n "RPi Bluetooth".
Then I can search for a new bluetooth device and connect with rpi bluetooth.

When I restart everything and want to connect rpi bluetooth again, I first have to delete rpi bluetooth in my laptop and search for a new bluetooth device.
Otherwise I can't find rpi bluetooth in synthesia.

I hope you can help me with this issue.
Thank you in advance and a happy 2020!

onlaj, 2019-12-31:
Hello.
Try to install Blueman (connect to your RPi with VNC), this software helped me a lot with managing bluetooth connections.

bradliegek3, 2019-12-31:
Will try this tomorrow, thankyou

bradliegek3, 2020-01-01:
Happy 2020 everyone.

I tried connecting with the help of blueman. The following occurs:

-I don't have to type anything in the console for RPi-Bluetooth to show up, so thats good. But, when I connect to RPi-Bluetooth with my laptop, I can not see RPi-Bluetooth in synthesia.

-If I connect to RPi-Bluetooth with my galaxy s8, I am able to see RPi-Bluetooth in synthesia. But, synthesia will not react to the piano keys being pressed.

Any idea's? (Yes I did all the synthesia steps to enable midi for windows and the dll file step.)

onlaj, 2020-01-07:
> -I don't have to type anything in the console for RPi-Bluetooth to show up, so thats good. But, when I connect to RPi-Bluetooth with my laptop, I can not see RPi-Bluetooth in synthesia.

So the connection is visible on your PC? I had similar problem and it was because Windows thought that RPi BT is a headphone device. Reconnecting solved it.

bradliegek3, 2020-01-12:
The connection is visible in a way. Windows can only see "unknown device". When I connect to that, I get 2 extra devices. One called Bluetooth LE Device .... and the other one RPi Bluetooth. It doesn't connect with RPi Bluetooth, but it only says paired. I can't connect with RPu bluetooth, only remove device. The other device is connected however and I can see my laptop is connectet with the RPi when I look at it via VNC. This does not work for synthesia however. Nothing shows up.

[image]

bradliegek3, 2020-01-14:
> Hello.
> Try to install Blueman (connect to your RPi with VNC), this software helped me a lot with managing bluetooth connections.

With the help of blueman I am able to disconnect Bluetooth LE Device B827eb5d4445 and connect with RPi bluetooth.

Blueman calls the bluetooth LE device "handsfree" and calls RPi bluetooth "audio source". In windows you can see the little audio icon next to RPi-Bluetooth.

I do not know how to adjust this or how to fix this. As of this moment I am still not able to see RPi-bluetooth in synthsia.

onlaj, 2020-01-25:
This is how connecting looks on my setup, I'm not sure how else I can help you
https://streamable.com/751sv
UI is not in english but you should figure it out.
Remember that you have to initiate connection from your PC, not the other way.

bradliegek3, 2020-01-25:
Interesting. Does your RPi-bluetooth jump to the audio list by itself? By me it stays in the other devices list and stays on "paired" instead of connected :o

onlaj, 2020-01-25:
Yeah, as you can see on video.
Try to wait for a while between each action. I cut some parts of video to make it shorter so you can't see, but I waited like 15 sec while it was searching for device, then another 15 after clicking "connect" etc.

bradliegek3, 2020-01-25:
I will have another try at this tomorrow, thank you for replying.

The only thing odd still, is that it connects to that other "weird" bluetooth device simultaneously with RPi-Bluetooth.

bradliegek3, 2020-01-26:
No idea what is happening with that stupid device anymore.
Before installing the bluetooth part, all led's work fine. After installing the bluetooth part, the led's glitch.
After installing the bluetooth part, no device is being found in windows. Only after I type sudo btmidi-server -v -n "RPi Bluetooth" in the console. I can find RPi-bluetooth in windows and synthesia recognizes it.

My hands are tied, no idea what to do. Every time I try it, I get another problem.

vzoltan, 2020-02-20:
@bradliegek3 
so, how did you solve that part that synthesia now sees your rpi?
bc for me synthesia doesnt see my pi, im guessing bc windows connects to the pi as headset/audio device, not midi.

bradliegek3, 2020-02-23:
@vzoltan  
Exactly as I wrote in that comment. I cannot get to get it working without typing that line.

I have not tried anything for a while because I moved to another house.
I will post again when I get any more progress.

randaz81, 2020-02-27:
I have more or less the same questions.
I reached this point so far:

[image]

I have two (can you explain me why two?) bluetooth devices displayed on windows, one is "connected", the other one is just "paired".

However, Synthesia is not able to detect the keyboard via bluethooth, even following the trick https://synthesiagame.com/forum/viewtopic.php?p=46920#p46920

The leds are working OK when I press the keys on my keyboard.
What else should I check?

vzoltan, 2020-02-27:
You should name your bt midi server service different to see if you are connected to it or to the pi.

Also in the btmidi.service file set the sleep to 30sec.

Reboot and it should work.

bradliegek3, 2020-02-28:
Unbelievable, it works!

closed 2020-02-28 by bradliegek3