onlaj/Piano-LED-Visualizer #59: ImportError: No module named webcolors

hyuny223, 2020-01-04:
when i connect my rasp to PC and type "sudo python -E visualizer.py", this message pop up.

Traceback (most recent call last):
  File "visualizer.py", line 15, in <module>
    import webcolors as wc
ImportError: No module named webcolors

I already install webcolors using "pip install webcolors".
what should i do??

hyuny223, 2020-01-05:
plus, I had "keyboard interrupt" error, so when I turn on my raspi, keyboard does not work.

JAKJUS7, 2020-01-05:
I am having the same trouble just getting one of the things to install. How is your piano/ keyboard hooked up to your pi or computer cause that is the part I'm stuck on. Does it connect directly to the pi or through the pc?

onlaj, 2020-01-05:
Try to install this library manually
https://webcolors.readthedocs.io/en/1.5/install.html
As for KeyboardInterrupt error, there is nothing in the code to listen for key press, I have no idea why it is happening. There is a chance that these problems are caused by your linux distribution, I'm testing everything on Raspbian.

@JAKJUS7 You connect your piano directly to RPi (you probably need OTG cable, normal usb -> micro usb), PC is only needed if you want to use it with Synthesia.

JAKJUS7, 2020-01-11:
I tried to manually install it, but it still gives the same error. Would it consider the LED strip as the module? Because I had it working a little bit ago, but I can not get it to stop giving me the error.

Traceback (most recent call last):
  File "visualizer.py", line 15, in <module>
    import webcolors as wc
ImportError: No module named webcolors

KapJI, 2020-05-14:
You actually needed to install it for root with `sudo pip install -r requirements.txt`.

closed 2020-12-23 by onlaj