onlaj/Piano-LED-Visualizer #60: Rasberry Pi0 delay and some keys dont respond

Saurabh528, 2020-01-05:
Hey man! 
I initially had rpi3 and had this error of delays in the key press as well as some keys dont respond.
Should I change the midi usb cable?
Also I am not using a Power adapter for ws2812b. Direct power supply to rpi and from rpi to lights. 
Can this also cause delay? Thanks :)

onlaj, 2020-01-06:
Hey.
Someone on youtube wrote:

> i tried Raspberry 3B and the latency is very obvious, then I switched to Raspberry Zero and the latency has gone. see the links for the difference. the Version of Raspberry Pi does matters.
> zero version https://youtu.be/KLxL-75BBDM
> 3B version https://youtu.be/zcuDTIgvlag

I don't think direct power from RPi can cause delay but I would avoid lighting more than few LEDs at the same time. RPi alone is not powerful enough to light up all leds, don't run any led animations.

Saurabh528, 2020-01-06:
Thanks I will try and let you know

Saurabh528, 2020-01-06:
Can the light type also cause delay I ordered 60 LEDs per metre by mistake. Is it causing delay

onlaj, 2020-01-07:
No, I'm pretty sure it is caused because you used RPi 3 instead of Zero, many users reported this problem.

Saurabh528, 2020-01-16:
Thank you so much it works perfectly. 
Another doubt is that the 1'44 lcd hat is not available in my region.

https://robu.in/product/3-2-inch-tft-lcd-screen-for-raspberry-pi 
Will this LCD be compatible with the code you have written. I will be able to change colors then :)

Saurabh528, 2020-01-16:
And please tell me How can I change the color using the code only. Which part I need to edit. I am comfortable doing it thanks a lot

Saurabh528, 2020-01-16:
Thank you so much buddy 
and LCD hat not available through your link 
https://www.banggood.in/1_44inch-128x128-Pixels-SPI-Interface-LCD-Display-HAT-for-Raspberry-Pi-p-1285050.html?cur_warehouse=CN
Is this fine for use

closed 2020-01-16 by Saurabh528