onlaj/Piano-LED-Visualizer #61: mido input list

hyuny223, 2020-01-06:
I wrote that LEDs do not respond to key press.
I found that my piano name is not on the midi input list.
But screen saver works well(the screen saver stops if it is not connected to the piano).
any solution?? :(

onlaj, 2020-01-06:
What linux distribution do you have installed? On your screens I can see Debian, is it right? Maybe try to install Raspbian?

hyuny223, 2020-01-06:
you are right. I installed this.

onlaj, 2020-01-07:
Type
`aconnect -l`
in your console to check if your piano is visible on the list. If it's not I'm afraid I can't help you. This might be both hardware or software problem but not related to my code. Maybe your piano requires some drivers when using with Linux.

hyuny223, 2020-01-07:
then it isn't related to whether debian or noobs?
very sad not to work.. thanks for your effort..!

closed 2020-12-23 by onlaj