onlaj/Piano-LED-Visualizer #62: No playback port & leds doesnt respond to keys

Roloo38, 2020-01-07:


onlaj, 2020-01-07:
Don't worry about playback port.
I can see that your keyboard is seen by system as two devices. Try to change input in Visualizer menu (Port Settings -> Input) to something else.

Roloo38, 2020-01-07:
@onlaj Thanks for the fast replay!! 
I’m a beginner at raspberry, so my question is: how do I change the input in visualizer?

 I dont use the little waveshare screen, i have the raspberry connected to a monitor with hdmi, keyboard and mouse.

onlaj, 2020-01-07:
Oh, unfortunately you can't change any settings without screen hat, you would have to edit the code and set input as a fix name, then try all available ports.

Roloo38, 2020-01-07:
And if i connect the led screen, i can change it directly?

Because i have the screen, but i didnt use it yet

onlaj, 2020-01-07:
Yes, with screen connected you will have menu with all settings.

Roloo38, 2020-01-07:
I connected the screen, and is white, nothing appears on it. So i was followed the instructions from “ https://neuma.studio/rpi-as-midi-host.html “, and i got some problems with “lcd-show.py” and with 

> lcd-show.py "first line" "second line" "here's a quite long line" "4th line of text"

> connectall.rb

Any ideea why?

[image]
[image]

onlaj, 2020-01-07:
You have to run visualizer.py to see menu.

Roloo38, 2020-01-07:
I runned the visualizer.py and the screen is still white. I think i didnt set it up correctly. I followed the instructions on the neuma.studio website and i got an error with “lcd-show” which cannot be found
Any ideea why?

onlaj, 2020-01-07:
You should only follow those parts of neuma tutorial:

- configuring automatic midi connection (this is required even if you don't need Bluetooth feature)
- midi bluetooth setup

Nothing more.

Roloo38, 2020-01-07:
And for the screen is some code i need to run or just plug and play? Because the screen still doesn’t work even when i run the visualizer.py...

onlaj, 2020-01-07:
If you connected it correctly it should work. Can you take a picture?

Roloo38, 2020-01-07:
So I rebooted it and now work (sorry, my fault:)

I’m gonna connect the cables for leds, and let you know if it works

Roloo38, 2020-01-07:
All is okay, working well. Thanks for your help and time!!

closed 2020-01-07 by Roloo38