onlaj/Piano-LED-Visualizer #63: configure midi connection

Dilanka-lakshan, 2020-01-08:
i cant configure midi connection using telnet. the interface of coding is like a mess and cant copy paste the codes and etc. what can i do
[image]

https://www.youtube.com/watch?v=-I_73CZXyI8&t=76s  .................... i followed this youtube tutorial from ap piano

onlaj, 2020-01-09:
Use SSH instead
https://www.putty.org/

Dilanka-lakshan, 2020-01-10:
thak you. i finally manage to finish that but . after soldering wires to rp and to lcd the item is not powering p when using power adapter to power

Dilanka-lakshan, 2020-01-10:
using dc power socket with the dc5v and 5a power adapter. why is that

closed 2020-12-23 by onlaj