onlaj/Piano-LED-Visualizer #64: lcd hat is white color display and nothing showing in it

Dilanka-lakshan, 2020-01-11:
after connected to power rp is working and lcd also work. but lcd hat is showing white display nothing showing in it

onlaj, 2020-01-11:
Yep, that's how it should work. You need to run visualizer.py to see menu.

closed 2020-01-13 by onlaj

Mihairu023, 2020-02-04:
Had the same problem on PRC's RPi 3 model B. Switching to RPi Zero WH fixed that as well as delays. 
(tip: if display being white even **after** launching visualizer.py is the issue then using different RPi may help)

Dilanka-lakshan, 2020-02-05:
do i need to buy a another pi to do my works. it is still the white. i
programmed the whole pi by formatting the sd card and coded. but it stil
white

Mihairu023, 2020-02-05:
Depends on which pi you have now. The best one is [Zero WH](https://www.distrelec.de/Web/WebShopImages/landscape_large/5-/01/Raspberry_Pi_Zero_W_30113445-01.jpg).

Dilanka-lakshan, 2020-02-05:
mi pi is zero wh . what is the reason for that

Mihairu023, 2020-02-05:
Good then, because that 1.44 screen required is designed specifically for Zero WH. (and no delays too). Did you put the lcd hat over pins?

Dilanka-lakshan, 2020-02-05:
yeah i put but it kind of a side angle view. i ll send you a photo

Dilanka-lakshan, 2020-02-08:
yeah i put it. but why is still white color

Mihairu023, 2020-02-08:
It will show the menu after you launch the "visualizer.py" file, not earlier. Did you launch it?