onlaj/Piano-LED-Visualizer #65: configuration for 49-key midi controller

Tallis161, 2020-01-11:
So I've tried just changing the led count to 98, as well as a few lines under keylist, keylist_status

The resulting behaviour is on the default octave the light activated is an octave too high, as well as being maybe 5 leds too high.
I had a look through the config but I don't see a way to tell the software the first led is the first key of a 49 key controller, and to ignore octave change messages.
Awesome to see the three led option implemented btw <3

onlaj, 2020-01-11:
Search for a line commented with
`#changing offset to adjust the distance between the LEDs to the key spacing`
In future updates I will probably add an option to change offset directly in Visualizer menu.

Tallis161, 2020-01-11:
ok, I'll try playing with that, though isn't the offset between the keys different to offset from the beggining of the strip?

Pianoman871, 2020-04-22:
Q: How can I change the note offset from the beginning of the strip for an 61 keys keyboard?

onlaj, 2020-05-14:
@Pianoman871 @Tallis161 https://github.com/onlaj/Piano-LED-Visualizer/issues/119

closed 2020-05-14 by onlaj