onlaj/Piano-LED-Visualizer #67: Standoff and screws too short

baksteen-13, 2020-01-14:
I built the project and everything is working great. now when assembling it inside the 3D printed case i find that the screws are too short to poke through the plastic + LCD hat to screw into the standoff. and on the bottom the standoffs aren't long enough to pass through to the bottom. Where did you aquire the standoffs and screws you used and what size are they?

onlaj, 2020-01-14:
I just found some long screws and cut them to the right length. I didn't specify it in instruction, sorry about that.
As for standoffs (to keep screen from bending), I used some plastic scraps, I think from a pen or something.
I will try to make more photos later.

The-Torrent, 2020-02-08:
> I just found some long screws and cut them to the right length. I didn't specify it in instruction, sorry about that.
> As for standoffs (to keep screen from bending), I used some plastic scraps, I think from a pen or something.
> I will try to make more photos later.

sorry to annoy but did you ever get a chance to get those photos?

i'm having issues tryna figure out where the wires connect when the lcd is placed over the pins.

baksteen-13, 2020-02-08:
You mean on the pins? maybe this video will help. this guy follows this project and made a long video about it.
https://youtu.be/-I_73CZXyI8?t=1888

vzoltan, 2020-02-18:
@baksteen-13 hi
Could you pls make a copy of your sd card, like and img file.

I just cant make it work, i tried many times from zero, but i never was able to run vizualizer.py, always fails the same step during the setup process (steps).

closed 2020-12-23 by onlaj