onlaj/Piano-LED-Visualizer #68: Synthesia learning with USB

bradliegek3, 2020-01-18:
Hey Onlaj,

Since I can't get the bluetooth part working after a few weeks of frustration. Is it possible to use synthesia and the led's if I connect my Piano, RPi and laptop to a usb hub or something?

bradliegek3, 2020-01-23:
I tried it. Didn't work

onlaj, 2020-01-24:
You need midi splitter, but don't ask me how to connect them because I never did that.
I tried connecting it this way https://imgur.com/bKxKGUN and in theory it should work, but I bought some cheap midi to usb interfaces and they didn't work at all with my piano. With good quality interface it should work fine, but it is getting expensive there.
Something like this should also do the job https://compasflamenco.com/midi-c-3/midi-usbusb-p-4.html but I don't want to recommend it because I didn't test it yet and shop supply is pretty low (2 units left at the moment).

bradliegek3, 2020-01-24:
Thanks for the reply.
My piano has a usb-b type output instead of the dual midi output.
Is it better achieveable that way?
And if I get it working, will the piano leds react to synthesia in theory?

onlaj, 2020-01-24:
> My piano has a usb-b type output instead of the dual midi output.
It doesn't change anything because at the end we are connecting piano to RPi with usb.
With this Sevilla USB-USB thing I'm 90% sure that it will work.
The only problem is that if I'll suggest it as official solution this shop will run out of them very quickly. I think this is a niche product, I never saw a similar alternative.

bradliegek3, 2020-01-25:
Sorry, I do not see how this would work. When connecting usb to the usb thingy and then again usb thingy to usb pc.

But if I get the bluetooth part working, I will not need this solution. So can you get a look at my bluetooth post please? #57

closed 2020-01-26 by bradliegek3