onlaj/Piano-LED-Visualizer #72: How to update to a newer version?

sergiusmuzzz, 2020-01-31:
I installed visualizer just a bit more than one month ago and I've seen at least a couple of good features were added to the project. I wonder if it's possible to update without breaking all preliminary  setup that I had to do before installing visualizer itself. And weather I will need to set up auto-run again after update even if I've done it first time. 

Is it as easy as just re-cloning git repo to PI and everything will work or there are some other necessary steps?

Additional info  in readme on how to update would be a good thing to have.

onlaj, 2020-01-31:
You just have to type
`git pull origin master`
in repository folder.
Nothing _should_  go wrong.

sergiusmuzzz, 2020-01-31:
Everything works after git pull. Easy peasy 👍

closed 2020-01-31 by sergiusmuzzz