onlaj/Piano-LED-Visualizer #74: Sequences. Can't add less than three colors if step is multicolor.

sergiusmuzzz, 2020-01-31:
I could add just two colors and it worked fine In version that I pulled on Dec 16, 2019.
This worked before update:
`<step_6>
        <color>Multicolor</color>
        <color_1>255, 255, 255</color_1>
        <color_2>255, 55 , 0</color_2>
        <light_mode>Normal</light_mode>
</step_6>`   

In latest version however if to go to multicolor step with two colors pi freezes and won't work until restarted.
To make it work with two colors I need to duplicate both.
`<step_6>
        <color>Multicolor</color>
        <color_1>255, 255, 255</color_1>
        <color_2>255, 55 , 0</color_2>
        <color_3>255, 255, 255</color_3>
        <color_4>255, 55 , 0</color_4>
        <light_mode>Normal</light_mode>
</step_6>`

onlaj, 2020-01-31:
I forgot to add support for multicolor range in sequence mode. Fixed now.
Make a backup of your sequences.xml file because it will be replaced with new one after you pull changes.
This is how it should looks now:
```
<color>Multicolor</color>
<color_1>210, 40 , 40</color_1>
<color_range_1>0, 255</color_range_1>
<color_2>0, 255, 80</color_2>
<color_range_2>0, 255</color_range_2>
```

sergiusmuzzz, 2020-01-31:
It's fixed. Thanks a lot!

closed 2020-01-31 by sergiusmuzzz