onlaj/Piano-LED-Visualizer #75: No module named webcolors

Crazy-Geeks, 2020-02-02:
Hello. I have problem with installed webcolors, visualizer.py don't want to work
In python IDLE it works fine
[image]

onlaj, 2020-02-02:
Hi.
Correct command to launch script is
`sudo -E python visualizer.py`

Crazy-Geeks, 2020-02-02:
I have just installed all libs by apt-get or from github (like apt-get install python-webcolors)
And after installing all the packets, my command works fine. Does it matter where the [-option] is?
I'll try correct command, but later
I used lite version of raspbian, it was needed to made 4GB distribution, instead of 9-10 GB

[image]

[image]

closed 2020-12-23 by onlaj