onlaj/Piano-LED-Visualizer #76: Raspberry pi 4?

PIrKo, 2020-02-04:
Hi, I can't get the zero w model. Anyone try on a rp4? and a touch screen? Thanks a lot

Kadnium, 2020-02-04:
Rasperry pi 3 has issues with leds being delayed so rp4 might have the same problem. I had rp3 and had to buy zero because of that.

PIrKo, 2020-02-06:
Thx i could buy the zero on aliexpress! thx

spiccrix, 2020-04-27:
Quick note! I have the rpi4 (4gb Ram). When I launch the visualiser.py with python (2) it lags. When I do it with python3 it works wonders!! So, I knew it from the beginning, it is not acceptable that rpi4 couldn't handle a basic script perfectly. More, I run this script with two other scripts controlling 2 other led strips, and it still works normally with no lag. I hope all rpi4 owners see this and try this solution.

spiccrix, 2020-04-27:
PS : to be able to run it with python3, you should install all needed python packages with "pip3" instruction

vzoltan, 2020-04-27:
Hi, how fast is it to load a midi file?

closed 2020-12-23 by onlaj