onlaj/Piano-LED-Visualizer #79: Feature request - Configurable timing for Finger-based channel

vzoltan, 2020-02-21:
I try to explain :)

id like to get the visual information about the next notes that has to be played x-sec earlier than it supposed to played.

benefit: i could prepare myself to play the next notes, 
like synhtesia gives the visual feedback of the following notes in its animation.
(dont get confused, i m not asking for any animation, just to be able to see the following notes earlier as the actual play time)

onlaj, 2020-02-21:
You will have to ask Synthesia's author for that feature :)

vzoltan, 2020-02-21:
i thought you visualizer sees the upcoming notes ahead not just right then.
or how does it work at your end?

onlaj, 2020-02-21:
Synthesia is sending upcoming notes with proper channel number at the moment you are suppose to play them, not earlier.

vzoltan, 2020-02-21:
so we need synthesia to send the animation data too, and then you could work with that?

onlaj, 2020-02-21:
We need Synthesia to send midi data about upcoming notes earlier, as soon as possible would be the best.

vzoltan, 2020-02-21:
Ok, i linked this feature request on their forum, i hope they will click on the link and come over here to take a look ;)
Thanks

npiegdon, 2020-02-25:
Hmm, this may be a bug in Synthesia.  In older versions (at least in Rhythm and Recital modes, but I thought it was Melody too), it would send the key light message as soon as you're inside the timing window (which is adjustable and defaults to about a half second).

Without that, the key lights are a lot less useful in Rhythm/Recital. :/

For Melody it's not so bad.  The only time you should need to look away from the screen is when you weren't able to find something in time... which means the keys will already be lit.

----

By the way, this is a cool project. :D  While I'm here, it looks like your readme.md is at least a little out of date.  Here's the official status of Bluetooth support in Synthesia:
- BLE MIDI on macOS: completely automatic and supported
- BLE MIDI on iOS: completely automatic and supported
- BLE MIDI on Win10: enable the "Midi.UseWinRTMidi" advanced option to try and use Microsoft's [complete mess](https://www.synthesiagame.com/forum/viewtopic.php?p=47530#p47530) of a UWP driver.
- BLE MIDI on Android: if your device supports the "Android M MIDI" feature, just connect to the MIDI device using [these instructions](https://synthesiagame.com/forum/viewtopic.php?p=47541#p47541) and it should "work" fine, with all of Android's awful latency and dropped events.

onlaj, 2020-02-25:
Yeah, I'm aware of that, but where did you find this list? Is it official? 
I will update readme soon.

npiegdon, 2020-02-25:
I'm the author of Synthesia. 😄

I'm not very consistent at writing documentation, but that is the current state as of today / version 10.6.  (This is probably the most concisely it has been written in a while.  I probably ought to get this on the site or in the forums someplace.) 😆

onlaj, 2020-02-25:
Oh, great so it's official then :D, mind if I copy-paste it into my readme?

npiegdon, 2020-02-25:
That would be fine, hehe.

ryandotnet, 2026-08-28:
Hi @npiegdon 

Sorry for reviving this old issue - did you ever manage to resolve the bug, or add the ability to change how early the keylights are sent out?

Thanks!

npiegdon, 2026-09-04:
Honestly, after six years my memory is having a little trouble dredging up the context around your question... 😅

If I understand correctly, I think you're looking for the "Gameplay.NoteWindowUs" advanced setting.  (Hold your Shift key while launching the app to show the advanced configuration window... on desktop at least; there isn't a way to change these on iPad or Android yet.)

The default is 400,000 microseconds (or 400ms).  You should be able to adjust that number upward to show the key lights earlier than that.