onlaj/Piano-LED-Visualizer #80: Bluetooth

eengamer, 2020-02-21:
Todaz i set up mz raspberrz pi zero and forgot to setup the bleuthoot the first time after i got everything installed i realised mz mistake i added it afterwards but is doesnt show up on my phone how can i fix this?

vzoltan, 2020-02-24:
1. first check in this app, if it sees, connect to it in it, then in the phone's bt settings.
https://play.google.com/store/apps/details?id=com.mobileer.example.midibtlepairing&hl=en

2. if this app cant see it, try to reset bluetooth in the visualizer menu on the pi, and then try step 1.

eengamer, 2020-02-24:
I have done both steps still doesnt show up :/

vzoltan, 2020-02-24:
ssh into you pi and check if BTMIDI server is up and running.
sudo systemctl status btmidi.service

now im almost sure, youll get an error.

eengamer, 2020-02-24:
And what to do if it says error??

vzoltan, 2020-02-24:
lets see what it says..

eengamer, 2020-02-24:
Can we contact anywhere or do we do it via github

vzoltan, 2020-02-24:
paste it here.

eengamer, 2020-02-24:
[image]

eengamer, 2020-02-24:
Cant seem to find a error i think

vzoltan, 2020-02-24:
one more thing to try on widnwos:
https://www.microsoft.com/en-us/p/midiberry/9n39720h2m05?activetab=pivot:overviewtab

if you see your BTMIDI in this on windwos, while on windows you see Connected to pi, 
should work.

(as for my, i only can use synthesia, when this midiberry makes the connection.)

eengamer, 2020-02-24:
[image]
Didnt find anything

vzoltan, 2020-02-24:
ok, its clear that you dont have connection to your pi via bt.
idk :(

try to do the
OPTIONAL MIDI BLUETOOTH SETUP at here:
https://neuma.studio/rpi-as-midi-host.html
again....

eengamer, 2020-02-24:
[image]
I try reinsyreinstalling and this shows up

vzoltan, 2020-02-24:
ich spreche nur ein bisschen deutsch ;) aber ich deneke du need to install other things too;)

maybe sudo apt install gcc
and then again...

vzoltan, 2020-02-24:
what version of raspbian did you install on your sd?

eengamer, 2020-02-24:
Could we get contact on discord if that command doesnt wirk since i tried alot of things but havent got it working yet

eengamer, 2020-02-24:
> what version of raspbian did you install on your sd?

The latest version

eengamer, 2020-02-24:
[image]
WhoWhole time it says libudev is required but i cant install it im getting errors

eengamer, 2020-02-24:
[image]

closed 2020-03-16 by eengamer