onlaj/Piano-LED-Visualizer #81: OSError: Cannot open resource

rickstaa, 2020-02-23:
Just for documentation when you get an OSError while trying to open the CPU information on the raspberry pi you have to install the `sudo apt-get install fonts-freefont-ttf` package.

```
Traceback (most recent call last):
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 1342, in <module>
    menu.enter_menu()
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 575, in enter_menu
    menu.change_settings(self.current_choice, self.currentlocation)
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 730, in change_settings
    screensaver()
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 884, in screensaver
    menu.render_screensaver(hour, date, cpu_usage, round(cpu_average,1), ram_usage, temp, cpu_chart)
  File "/home/pi/Piano-LED-Visualizer/visualizer.py", line 599, in render_screensaver
    fonthour = ImageFont.truetype('/usr/share/fonts/truetype/freefont/FreeSansBold.ttf', 31)
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/ImageFont.py", line 642, in truetype
    return freetype(font)
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/ImageFont.py", line 639, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/ImageFont.py", line 188, in __init__
    font, size, index, encoding, layout_engine=layout_engine
OSError: cannot open resource
```

closed 2020-02-23 by rickstaa

onlaj, 2020-02-23:
Thanks, I will add it to instruction.

rickstaa, 2020-02-23:
@onlaj No problem. Thak you for the amazing package :)!

stepir, 2020-05-26:
> Thanks, I will add it to instruction.

just a gentle reminder @onlaj

rickstaa, 2020-05-29:
@stepir your right that's a typo. Thanks, I updated my comment for future reference.