onlaj/Piano-LED-Visualizer #85: Synthesia colors mixing with RPi colors

bradliegek3, 2020-02-28:
When I play a song on synthesia. The colors displayed are not always green and blue for left and right.
When I have my led color on red for example, some notes are displayed the right green or blue, but some will be red.
I have set input and playback to RPi instead of the piano.
How do I fix this?

vzoltan, 2020-02-28:
Same here.
It should be blue and green, but randomly the light up as default color from visualizer.

onlaj, 2020-02-28:
Do you have this problem when playing with "practice the melody/rhytm" or "Song Recital"?
I know it happens with "watch and listen only" but as far as I know it's Synthesia's fault. It's sending midi messages twice for each note. one with proper channel number and one without.

bradliegek3, 2020-02-28:
It happens with practice melody. Have not tried the others

onlaj, 2020-02-28:
Try to change some Synthesia settings.
For example, in BT Midi out I have:
My notes -> Disabled
Background Instruments -> Enabled
Key Lights -> Finger-based channel
Prevent "local" notes -> Enabled
MIDI Channel -> Omni (All Channels)

vzoltan, 2020-02-28:
Listening only mode
Nit touching the keys
And default color still lights up

bradliegek3, 2020-02-28:
I have the same settings. Problem still there

onlaj, 2020-02-28:
Do you have latest Synthesia installed?
Maybe try different MIDIs?

bradliegek3, 2020-02-28:
Latest synthesia is installed.

Tried all the midi channels:

- channel 1-11 visualizer color
- channel 12 blue
- channel 13 green
- channel 14-16 visualizer color

npiegdon, 2020-02-29:
> ... as far as I know it's Synthesia's fault. It's sending midi messages twice for each note. one with proper channel number and one without.

This is news to me.  Background Instruments should be set to Off if you're just interested in key lights.  That might explain the double-notes in Watch and Listen Only mode.  (One for the key light, one for the background instrument.)

bradliegek3, 2020-03-01:
Tried it with instruments off.
Restarted the pi a couple times.

No luck yet ;(

onlaj, 2020-03-01:
> This is news to me. Background Instruments should be set to Off if you're just interested in key lights. That might explain the double-notes in Watch and Listen Only mode. (One for the key light, one for the background instrument.)

Hmm, yeah, I tested it with Background Instruments set to OFF and now it works fine (no mixed colors), the problem is that there is no sound coming from my piano.
Video: https://streamable.com/9s84l

For me it looks like my piano fault, because my script have no problems understading midi messages coming from Synthesia.
A year ago I wrote to you by email that some midi files played by Synthesia do not play any sound on my piano, I managed to fix that by removing some "Control Change" messages from midi. This might be something similar.

@bradliegek3 you still have mixed colors even after you disabled Background Instruments? What about other options like "prevent local notes" or "my notes", did you try to change them (although I am not sure what they are for)?

bradliegek3, 2020-03-01:
Yes I still have mixed colors after setting background instruments to off...
I have tried all the settings, also tried the exact same settings as in your video. Still mixed colors.

vzoltan, 2020-03-01:
Hmm, yeah, I tested it with Background Instruments set to OFF and now it works fine (no mixed colors), the problem is that there is no sound coming from my piano.
Video: https://streamable.com/9s84l

Same, when bg instruments turned off, no music from piano since bg music is off and just the light is "coming" thorugh .

when bg music is on (to be able to listening to the piece im practicing, i also have the visualizers default color with synthesias green and blue colors.

npiegdon, 2020-03-02:
So, just to clear up my understanding: this device is not just for outputting to the light strip, but also forwards MIDI messages (via BLE) to a connected keyboard?

In that case, the solution (according to the way the MIDI protocol is sort of intended) is to report the RPi as two MIDI devices: one for key lights and one for forwarding.

In Synthesia you'd set "Finger Based Channel" key lights for the light device (with no other outputs enabled) and for the pass-through device, you'd set whatever you want (the way you might if it were connected via a wire directly).  That would solve this problem and generally make things a lot more flexible.

onlaj, 2020-03-02:
@npiegdon 
> So, just to clear up my understanding: this device is not just for outputting to the light strip, but also forwards MIDI messages (via BLE) to a connected keyboard?

Yep, it works as passthrough between Piano and Synthesia, works in both directions
Piano <-> RPi <-> Synthesia
It just intercepts passing messages, we can choose from which direction. If we choose Piano it reacts to key press, if we choose BT service it reacts to messages coming from Synthesia or other software connected by bluetooth.

For me, when I have Background Instruments set to ON and play "Practice Melody" with both hands I see only blue and green colors, correctly matching the hands. When I practice one hand, let's say right one, it lights up next-to-play keys with green color, while left hand is played by itself with sound coming from my piano and keys are lighten up with white color (or different, depends on what color I choose).
So it works exactly as it should.
Video: https://streamable.com/wu10i (leds on left are white, white balance made it look like light blue)

The problem occurs when I launch "watch and listen only", then it tries to light up keys with both colors at the same time, blue/green and white.
Solution is to disable Key Lights in settings, then there is no blue/green color and sound is played correctly. 
I can also add an option in my script to ignore messages with/without key light parameter, it would work too.

Unfortunately @bradliegek3 have problem with mixed colors in practice mode and even with Background Instruments set to OFF which I can't reproduce on my setup. I have no idea what could be the reason.

vzoltan, 2020-03-02:
> @npiegdon
>
>
> Yep, it works as passthrough between Piano and Synthesia, works in both directions
> Piano <-> RPi <-> Synthesia
> It just intercepts passing messages, we can choose from which direction. If [...]

(Then why cant i get it work with android, just with windows? just asking. -android works with synthesia, usb or bt, both. )

onlaj, 2020-03-02:
> (Then why cant i get it work with android, just with windows? just asking. -android works with synthesia, usb or bt, both. )

¯\\\_(ツ)\_/¯
I can't even find Synthesia on play store now.

vzoltan, 2020-03-02:
i just sent you the latest apk from my phone.

npiegdon, 2020-03-02:
This is disconcerting.  My internal Google Play admin page is showing that everything is fine and that the app status is "Published", but the usual link to the download page is showing a "page not found" error.  How bizarre.  Thanks Google...  *sigh*

bradliegek3, 2020-03-02:
Maybe it's a good idea to make another topic about the whole Windows/android/ios issue.

Then we can track the led issues in this topic again :)

vzoltan, 2020-03-02:
dont worry im sure olaj and npiegdon will take care of this led issue, im also experiencing it.

bradliegek3, 2020-03-02:
I really appreciate the help :)
The whole platform issue maybe is something to look at in the future though.
For example, the RPi isn't even recognisable on an iPad.

onlaj, 2020-03-02:
Well, I tried with android using the instruction from [here ](https://synthesiagame.com/forum/viewtopic.php?p=47541#p47541)
BT device is visible, but doesn't react when I play piano.

[image]

vzoltan, 2020-03-03:
Here is my video to show, not just blue and green, but rhe default (this case white) white leds also light up.

bradliegek3, 2020-03-03:
You stil get a lot of green and blue.
I think I get more than 75% default color notes and only very little blue and green.

npiegdon, 2020-03-04:
In case it got lost earlier in the thread before that Google Play stuff, the "Right Thing" in this case is to have the RPi report itself as two different devices:

1. One for the light strip (only).
2. One that forwards your messages to any connected devices.  (Or maybe one *per* connected device.)

That would solve this problem outright.  You could send key light messages to the first device and background notes to the second.

The bug here is that there is no way to direct the messages to (only) the correct place.

This would also enable scenarios where you could disable the light strip temporarily and just use the RPi as a kind of BLE-to-MIDI adapter.  (Imagine someone nearby late at night that was bothered by the lights, but you still want to play with headphones, etc.)

vzoltan, 2020-03-04:
> Imagine someone nearby late at night that was bothered by the lights, but you still want to play with headphones, etc

(You can set the brightness lower ;) mine is set to 1% (for this video increased to 100) )

onlaj, 2020-03-04:
@npiegdon or just add an option to skip messages :D https://streamable.com/pr5mf
@bradliegek3 I committed changes, you can try if it works for you

bradliegek3, 2020-03-04:
I will try this afternoon.
Just pull the changes and see if things work?

onlaj, 2020-03-04:
Yep

npiegdon, 2020-03-04:
> ... or just add an option to skip messages :D

That works, too!  I'd still argue multiple devices would be more flexible/useful/correct in the long run, but there is no arguing with how much faster it is to implement a filter. 😛

Nice!

vzoltan, 2020-03-04:
> > ... or just add an option to skip messages :D
> 
> That works, too! I'd still argue multiple devices would be more flexible/useful/correct in the long run, but there is no arguing with how much faster it is to implement a filter. 😛
> 
> Nice!

so if this is solved;) do you have anything on the "issue" with android+led visualizer? (above)

bradliegek3, 2020-03-04:
When trying to pull the changes with git pull origin master I get the following:
error: Your local changes to the following would be overwritten by merge: settings.xml
Please commit your changes or stash them before you merge.

Any idea?

onlaj, 2020-03-04:
`git checkout settings.xml`

You will loose your settings, keep it in mind.

bradliegek3, 2020-03-04:
okay nice it works!
No more mixed colors, but only when background instruments is set to off. So that means no sound unless on watch and listen..

onlaj, 2020-03-04:
Wait, did you change settings under "LED Strip Settings -> Skipped notes"?

bradliegek3, 2020-03-04:
I can't even find the skipped notes setting so I guess not?

vzoltan, 2020-03-04:
you need to restart the pi, or the visualizer

onlaj, 2020-03-04:
Open menu.xml and check if there is "Skipped notes" submenu.

bradliegek3, 2020-03-04:
I do not have the skipped notes submenu. Either on the pi or if I look in menu.xml

onlaj, 2020-03-04:
So you didn't pull the changes, try again.

bradliegek3, 2020-03-04:
I can see it now, thanks

vzoltan, 2020-03-04:
i just tried the new version, well it solves this issue but:) i noticed something.

so, whitout this, what i liked that i got feedback when i hit the key in wrong timing bc when i hit out of sync, my color (red in my case) light up. when i was in sync, no other colors shown, just blue and green.

now with this new version, yes i can turn on or off the "glitch" but that also takes away this "feature" i liked during practice.

just saying, just an observation ;)

vzoltan, 2020-03-04:
(but it also takes away that feature, that this "glitch" gives:
to get a feedback when you hit a wrong key, or when you hit with wrong timing, out of snync.
i like, when i do a mistake, the led lights up in red (default color for me) and also i can see the color guidence as blue and green at the same time, its just an observation ;) )

closed 2020-10-11 by bradliegek3