onlaj/Piano-LED-Visualizer #89: Waveshare Installation/White Screen

xsound09, 2020-03-05:
Hey there,

Is there further installation required for the Waveshare screen other than the instructions here? My screen only shows white when plugged in, I have followed all the instructions here, and also the manufacturer's instructions (afterwards, since it isn't working).

I am wondering if my Waveshare screen is faulty, haven't seen a single graphic on it other than blank white. Any advice? Thanks!

onlaj, 2020-03-05:
It might be a stupid question, but did you launch visualizer.py script?

xsound09, 2020-03-07:
@onlaj Yes, nothing happens on the visualizer when launching the script. I'm mostly curious if there's outside installation required for the screen beside your instructions because the instructions the manufacturers provide have bad English so I'm wondering if I installed it incorrectly with their instructions, or if that's even relevant if I correctly follow yours.

onlaj, 2020-03-07:
What does the console say?

xsound09, 2020-03-20:
@onlaj My apologies for the delay. When I run sudo -E python visualizer.py the console says: 

Inport set to Keyboard:Keyboard MIDI 1 20:0
Inport set to Keyboard:Keyboard MIDI 2 20:1
playport set to Keyboard:Keyboard MIDI 1 20:0
playport set to Keyboard:Keyboard MIDI 2 20:1

But, nothing else happens. The waveshare screen remains white.

onlaj, 2020-03-22:
Try to run some example code to check if everything works correctly
https://www.waveshare.com/wiki/File:1.44inch-LCD-HAT-All-Code.7z

xhosan, 2020-04-27:
I am having the same problem as well. I checked the physical connections and everything seems okay, trying the example code now.

xsound09, 2020-04-29:
@xhosan How did the example code run for you? I haven't been able to check yet, since I have to disconnect my work monitor to test. Thanks for your help @onlaj I will be checking soon

xhosan, 2020-04-29:
@xsound09 I ran each of the programs in the file, still a blank white screen for all of them. Here's what was on my console for key_demo.py and main.py, they were the only programs that logged anything. Not sure if this helps or not but thanks @onlaj for everything!
`
pi@raspberrypi:~/RaspberryPi/python $ sudo -E python key_demo.py
key_demo.py:21: RuntimeWarning: No channels have been set up yet - nothing to clean up!  Try cleaning up at the end of your program instead!
  GPIO.cleanup()
^CTraceback (most recent call last):
  File "key_demo.py", line 100, in <module>
    disp.LCD_ShowImage(image,0,0)
  File "/home/pi/RaspberryPi/python/LCD_1in44.py", line 310, in LCD_ShowImage
    pix = pix.flatten().tolist()
KeyboardInterrupt
pi@raspberrypi:~/RaspberryPi/python $ sudo -E python main.py
**********Init LCD**********
***draw line
***draw rectangle
***draw text
pi@raspberrypi:~/RaspberryPi/python $
`

imagemas, 2020-06-02:
Help. mistakenly ordered another screen.
Waveshare 1.3inch OLED HAT, OLED дисплей 128×64px
How can i use it?
[image]

onlaj, 2020-06-02:
@imagemas you can't, at least not without making a lot of changes in code, sorry.

closed 2020-12-23 by onlaj